instantlinux/docker-tools #1: [NFR] Raspberry Pi Support for Weewx

ruudboon, 2018-01-11:
Would it be possible to add ARM/Raspberry pi support to the Weewx image?

instantlinux, 2018-01-12:
I do have an old Raspberry Pi here (which used to run my WeeWX setup until I moved it to Ubuntu): would have to update Raspbian and set up my existing Jenkins to trigger builds on it. I'll put it on my to-do list, I'll update here when I start to tackle this idea. Thanks for the suggestion!

ruudboon, 2018-01-12:
Cool, thnx

instantlinux, 2018-01-29:
I published versions 3.7.1 and 3.8.0 under Raspberri Pi. Look for the image [instantlinux/weewx-armhf](https://hub.docker.com/r/instantlinux/weewx-armhf).

instantlinux, 2018-01-30:
Had to make another fix to make this automatically build correctly; multi-arch builds are a pretty new thing for Docker so there aren't any instructions. It turns out to be fairly easy: create a hooks/pre_build script thus:
```
#!/bin/bash
# Set up QEMU for multi-arch build
docker run --rm --privileged multiarch/qemu-user-static:register --reset
```
If you're using a hooks/build script (to generate tags), it also needs to specify a dockerfile thus:
```
#!/bin/bash
DOCKERFILE_OPT=
if [[ $DOCKER_REPO == *-armhf ]]; then
  DOCKERFILE_OPT="-f Dockerfile.armhf"
fi
docker build -t $IMAGE_NAME $DOCKERFILE_OPT . \
  --build-arg VCS_REF=$(git rev-parse HEAD) \
  --build-arg BUILD_DATE=$(date -u +%Y-%m-%dT%H:%M:%SZ)
```
And to generate tags that match the application's version number, a hooks/post_push:
```
#!/bin/bash
TAG=$(grep "ARG WEEWX_VERSION" Dockerfile | cut -d= -f 2)
docker tag $IMAGE_NAME $DOCKER_REPO:$TAG
docker push $DOCKER_REPO:$TAG
```
I've added these to this repo so future versions of WeeWX get built properly.

closed 2018-02-01 by instantlinux