instantlinux/docker-tools #2: nut-upsd: Unable to use old-style MONITOR line without a username

martikainen87, 2018-01-17:
Hi

Downloaded your entire repository and tried to start the nut-upsd by running 'docker compose up' inside the nut-upsd directory after a quick change of the driver to use (blaser_usb)

It seems to work good but I get an error message that I cant track back to the dockerfile, compose.yaml or any of the other files inside the repository.

This is my compose.yaml

```bash
version: "3.1"

services:
  app:
    image: ${REGISTRY_URI:-instantlinux}/nut-upsd:latest
    restart: always
    environment:
      SERIAL: ${UPS_1_SERIAL}
      DRIVER: blazer_usb
    ports:
    - ${PORT_UPSD_1:-3493}:3493
    secrets:
    - nut-upsd-password
    privileged: true

secrets:
  nut-upsd-password:
    file: /var/adm/admin/secrets/nut-upsd-password
```

And this is the error message I get

```bash
pi@Dockerhost:/Docker/docker_compose/docker-tools/images/nut-upsd$ sudo docker-compose up
WARNING: The UPS_1_SERIAL variable is not set. Defaulting to a blank string.
Removing nutupsd_app_1
Recreating d26226d2ae10_nutupsd_app_1 ... done
Attaching to nutupsd_app_1
app_1  | cat: read error: Is a directory
app_1  | Network UPS Tools - UPS driver controller 3.6.0-5404-g047aab7fa3
app_1  | Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (3.6.0-5404-g047aab7fa3)
app_1  | Supported UPS detected with megatec protocol
app_1  | Vendor information unavailable
app_1  | No values provided for battery high/low voltages in ups.conf
app_1  |
app_1  | Using 'guestimation' (low: 20.800000, high: 26.000000)!
app_1  | Battery runtime will not be calculated (runtimecal not set)
app_1  | Network UPS Tools upsd 3.6.0-5404-g047aab7fa3
app_1  | fopen /var/run/nut/upsd.pid: No such file or directory
app_1  | listening on 0.0.0.0 port 3493
app_1  | Connected to UPS [ups]: blazer_usb-ups
app_1  |    0.000000    fopen /var/run/upsmon.pid: No such file or directory
app_1  | Network UPS Tools upsmon 3.6.0-5404-g047aab7fa3
app_1  |    0.000106    Using power down flag file /etc/killpower
app_1  |    0.000169    Unable to use old-style MONITOR line without a username
app_1  |    0.000196    Convert it and add a username to upsd.users - see the documentation
app_1  |    0.000217    Fatal error: unusable configuration
nutupsd_app_1 exited with code 1
```

instantlinux, 2018-03-07:
See the [README file](https://github.com/instantlinux/docker-tools/blob/master/images/nut-upsd/README.md):

*Note that the usbhid-ups driver for APC requires you to provide the correct 12-digit hardware serial number.*

closed 2018-03-07 by instantlinux

instantlinux, 2018-06-26:
The entrypoint script will create a /etc/nut/upsmon.conf file that contains this line:
```
MONITOR $NAME@localhost 1 $API_USER $API_PASSWORD $SERVER
```
The error message you posted here suggests that one of the above items is left blank. Since $NAME, $API_USER, and $SERVER are specified as defaults in Dockerfile, I believe you're missing the password. Did you put a value into the secret file listed in docker-compose.yml (/var/adm/admin/secrets/nut-upsd-password) and invoke _docker-compose up_ from a user with permissions to read that file? Also, double-check that the full contents of /etc/nut/upsmon.conf within the container look right; the directives in mine look like:
```
$ docker exec nutupsd_app_1 cat /etc/nut/upsmon.conf
# Network UPS Tools: example upsmon configuration
#
# This file contains passwords, so keep it secure.
#...
MONITOR ups@localhost 1 upsmon 7s[redacted]0 master
RUN_AS_USER nut
```

I have only tested this with my type of UPS so it's possible there's a driver compatibility issue. Here's my ups.conf:
```
[ups]
        driver = usbhid-ups
        port = auto
        serial = "3B16[redacted]5"
        desc = "UPS"
```

martikainen87, 2018-06-26:
Oh, i thought it was default password set to someting, is the nut-upsd-password file supposed to be at that location on my docker OS? or where do i place it?

instantlinux, 2018-06-26:
Create a file with a randomly-generated string and put it wherever you want, just make sure the docker-compose secret points to path of the file you've created.

martikainen87, 2018-06-26:
That was the key! Thanks for the help, and for anyone finding this thread in the future, this is what my files looks like

Dockerfile  - No changes from default

```
FROM alpine:3.7
MAINTAINER Rich Braun "docker@instantlinux.net"
ARG BUILD_DATE
ARG VCS_REF
LABEL org.label-schema.build-date=$BUILD_DATE \
    org.label-schema.license=GPL-2.0 \
    org.label-schema.name=nut-upsd \
    org.label-schema.vcs-ref=$VCS_REF \
    org.label-schema.vcs-url=https://github.com/instantlinux/docker-tools

ARG NUT_VERSION=2.7.4-r5
ENV API_USER=upsmon \
    DESCRIPTION=UPS \
    DRIVER=usbhid-ups \
    GROUP=nut \
    NAME=ups \
    PORT=auto \
    SECRET=nut-upsd-password \
    SERIAL= \
    SERVER=master \
    USER=nut

RUN echo '@edge http://dl-cdn.alpinelinux.org/alpine/edge/main' \
      >>/etc/apk/repositories && \
    echo '@testing http://dl-cdn.alpinelinux.org/alpine/edge/testing' \
      >>/etc/apk/repositories && \
    apk add --update nut@testing=$NUT_VERSION \
      libressl2.7-libcrypto@edge libressl2.7-libssl@edge

EXPOSE 3493
COPY entrypoint.sh /usr/local/bin/
ENTRYPOINT /usr/local/bin/entrypoint.sh
```

docker-compose.yml  - No serial added

```
docker-compose.yml

version: "3.1"

services:
  app:
    image: ${REGISTRY_URI:-instantlinux}/nut-upsd:latest
    restart: always
    environment:
      DRIVER: blazer_usb
    ports:
    - ${PORT_UPSD_1:-3493}:3493
    secrets:
    - nut-upsd-password
    privileged: true

secrets:
  nut-upsd-password:
    file: /Docker/nut-upsd/nut-upsd-password

```
I then created the nut-upsd-password in the same directory as the Dockerfile and that solved everything :)