instantlinux/docker-tools #3: Unable to run open-xchange-appsuite

jetwes, 2018-02-21:
I tried to run the ox appsuite docker file. It's always hanging at registering the server:
`server could not be registered:
Server response:
 Look-up failed. Service "OXUtil_V2" is not available. `
I'm getting the exact same message in my installation process of ox-suite without docker.. Any guess what's causing this? Couldn't find any related topic. It's strange. I tried it in ubuntu, redhat and now your docker-file...

instantlinux, 2018-03-08:
I assume this is a symptom of something else; look for any earlier error in your logs. More things to check:

* Make sure the database is set up properly
* Make sure the volume mounted as /ox/store has ownership open-xchange (UID 104)

The only other dependency I can think of is the Linux kernel version. I run under 4.11.12 but I'm guessing it can run on a 3.x kernel.

mgrundkoetter, 2018-08-21:
I have the same problem. Longer trace is this:
```
server could not be registered: 
Error: Connection refused to host: localhost; nested exception is: 
	java.net.ConnectException: Connection refused (Connection refused)
--Waiting on registerserver
server could not be registered: 
Error: Connection refused to host: localhost; nested exception is: 
	java.net.ConnectException: Connection refused (Connection refused)
--Waiting on registerserver
server could not be registered: 
Server response:
 Look-up failed. Service "OXUtil_V2" is not available.
	at sun.rmi.registry.RegistryImpl.lookup(RegistryImpl.java:227)
	at sun.rmi.registry.RegistryImpl_Skel.dispatch(RegistryImpl_Skel.java:115)
	at sun.rmi.server.UnicastServerRef.oldDispatch(UnicastServerRef.java:468)
	at sun.rmi.server.UnicastServerRef.dispatch(UnicastServerRef.java:300)
	at sun.rmi.transport.Transport$1.run(Transport.java:200)
	at sun.rmi.transport.Transport$1.run(Transport.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.Transport.serviceCall(Transport.java:196)
	at sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:573)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:834)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$0(TCPTransport.java:688)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:687)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
	at sun.rmi.transport.StreamRemoteCall.exceptionReceivedFromServer(StreamRemoteCall.java:283)
	at sun.rmi.transport.StreamRemoteCall.executeCall(StreamRemoteCall.java:260)
	at sun.rmi.server.UnicastRef.invoke(UnicastRef.java:375)
	at sun.rmi.registry.RegistryImpl_Stub.lookup(RegistryImpl_Stub.java:119)
	at java.rmi.Naming.lookup(Naming.java:101)
	at com.openexchange.admin.console.util.server.RegisterServer.<init>(RegisterServer.java:77)
	at com.openexchange.admin.console.util.server.RegisterServer.main(RegisterServer.java:91)

--Waiting on registerserver
```

My setup is just 2 docker containers in a network (no swarm). One mariadb and one oxserver. The mariadb is running fine, but I don't get the oxserver running... So I checked your hints (db and user rights) which are both fine. Any other idea?

rene-dekkers, 2018-08-25:
Same problem here.
For testing purpose I've created a OX Debian 9 image and the same error occurred. Any ideas?

rene@rene-MacBookPro:~$ docker exec -ti appsuite_ox_backend_1 bash
root@c26574c2cfab:/# /opt/open-xchange/sbin/listcon
listconfigdiff           listcontextsbydatabase   
listcontext              listcontextsbyfilestore  
root@c26574c2cfab:/# /opt/open-xchange/sbin/listcontext
contexts could not be listed: 
Server response:
 Look-up failed. Service "OXContext_V2" is not available.
	at sun.rmi.registry.RegistryImpl.lookup(RegistryImpl.java:227)
	at sun.rmi.registry.RegistryImpl_Skel.dispatch(RegistryImpl_Skel.java:115)
	at sun.rmi.server.UnicastServerRef.oldDispatch(UnicastServerRef.java:468)
	at sun.rmi.server.UnicastServerRef.dispatch(UnicastServerRef.java:300)
	at sun.rmi.transport.Transport$1.run(Transport.java:200)
	at sun.rmi.transport.Transport$1.run(Transport.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.Transport.serviceCall(Transport.java:196)
	at sun.rmi.transport.tcp.TCPTransport.handleMessages(TCPTransport.java:573)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run0(TCPTransport.java:834)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.lambda$run$0(TCPTransport.java:688)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.rmi.transport.tcp.TCPTransport$ConnectionHandler.run(TCPTransport.java:687)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
	at sun.rmi.transport.StreamRemoteCall.exceptionReceivedFromServer(StreamRemoteCall.java:283)
	at sun.rmi.transport.StreamRemoteCall.executeCall(StreamRemoteCall.java:260)
	at sun.rmi.server.UnicastRef.invoke(UnicastRef.java:375)
	at sun.rmi.registry.RegistryImpl_Stub.lookup(RegistryImpl_Stub.java:119)
	at java.rmi.Naming.lookup(Naming.java:101)
	at com.openexchange.admin.console.context.List.maincall(List.java:79)
	at com.openexchange.admin.console.context.ListCore.commonfunctions(ListCore.java:102)
	at com.openexchange.admin.console.context.List.<init>(List.java:70)
	at com.openexchange.admin.console.context.List.main(List.java:74)

leonmarkus, 2018-08-30:
Exactly the same issue here

instantlinux, 2018-11-08:
There are no requirements other than a blank data volume and a blank MySQL database with appropriate credentials. Please submit output of _docker log_ if there is any; I've never had this problem with the image so I have to assume the problem is related to content or permissions of the data volume or database.

closed 2019-01-13 by instantlinux