instantlinux/docker-tools #7: "Interrupted system call" on starting nut-upsd

Cyber1000, 2018-10-24:
Hi,

Thanks for the image nut-upsd, I wanted to add some changes and tried to build the image before making my changes. It worked, but when starting with docker-compose I'm getting the following (first line seems to be the problem):

> /usr/local/bin/entrypoint.sh: line 26: can't create /etc/nut/ups.conf: Interrupted system call
> Network UPS Tools - UPS driver controller 3.7.0-4118-g91b32f90c7
> Error: no UPS definitions found in ups.conf
> fopen /var/run/nut/upsd.pid: No such file or directory
> Network UPS Tools upsd 3.7.0-4118-g91b32f90c7
> listening on 0.0.0.0 port 3493
> Warning: no UPS definitions in ups.conf
> Fatal error: at least one UPS must be defined in ups.conf
> Network UPS Tools upsmon 3.7.0-4118-g91b32f90c7

When I look into the container a /etc/nut/ups.conf exists, but only the commented-out-part not the lines which should have been added.
The 2 other files get appended.

Any ideas?
Thanks!

Cyber1000, 2018-10-25:
After some testing it seems that it has nothing to do with my fork, I retested with this repository and the same docker-compose which is in this repository and getting the same issue.

But a restart mostly helps, so it seems to come occasionally, I have found a similar problem (https://github.com/Landoop/kafka-topics-ui/issues/77) which had the problem on the EOF line too, it seems like a kind of racecondition to me.

Sometimes it's only ups.conf, sometimes all conf-files in the entrypoint.sh, sometimes it works, but within five restarts I have at least one failure. 
I'm using:
- host: ubuntu 18.04 LTS
- docker: 18.06.1-ce
- docker-compose: 1.22.0

I assume this doesn't happen to you?

Cyber1000, 2018-10-25:
Had the same problem on a second computer (thought at first that it was the empty serial-number, cause my second computer has another UPS), but it was exactly the same with an older ubuntu host. Fxied it with some sort of template on my own image now (the entrypoint is quite different from this, so no easy merging possible)

If you see similar issues on your computer(s), feel free to ask. If nobody seeing the same issue, when executing "docker logs {yourdockerservice}" we may close this task.

Cyber1000, 2018-11-11:
Hi,
Did you have time to try?
As said I'm using my own image without EOF magic, but just to know if we should close this issue.

Greetings!

instantlinux, 2019-01-13:
Sorry for the delay looking at this: yes, I can reproduce similar symptoms by restarting a few times.

instantlinux, 2019-01-15:
I've pushed an update to the image 2.7.4-r6 at [instantlinux/nut-upsd|https://hub.docker.com/r/instantlinux/nut-upsd]. Try it and see if it fully addresses your symptoms.

nicxvan, 2019-01-19:
I was still getting this on latest. I switched to 2.7.4-r6 and the error went away. I still don't have it up and running cause it says there is no HID UPS but I will research that error.

nicxvan, 2019-01-19:
I might have spoken too soon, restarting a couple of times I still see it occasionally.

nicxvan, 2019-01-20:
It fails and just says it's stale, it is set to restart: unless-stopped, but the container doesn't die when it gets stale.

I will try latest and report back.

nicxvan, 2019-01-20:
Ok it's connected using latest: here is the docker-compose. I will keep an eye on it and let you know if it becomes stale and auto restarts.

I think I had to remove the serial number that seemed to help too.

```
  nutups:
    container_name: nut_ups
    restart: unless-stopped
#    build:
#      context: ./containers/nutupsd
    image: instantlinux/nut_upsd:latest
    environment:
      - vendorid=09ae
      - desc="Tripp Lite SMART1500LCD"
    network_mode: host
    ports:
      - 3493:3493
    devices:
      - /dev/usb/hiddev0:/dev/usb/hiddev0
    privileged: true
    secrets:
      - nut-upsd-password
secrets:
  nut-upsd-password:
    file: ./nutpass
```

nicxvan, 2019-01-20:
It still goes stale after a few minutes even on latest. Let me change from unless-stopped to always.

nicxvan, 2019-01-20:
Also I'm not sure if I should move this to a new ticket:
I think this container has a new error message:

```
   0.000000	fopen /var/run/upsmon.pid: No such file or directory
   0.000384	Using power down flag file /etc/killpower
```

instantlinux, 2019-01-20:
I think the /var/run/upsmon.pid message is harmless, it's caused by the /var/run directory's allowing write by root only (upsmon runs as user _nut_). But I'll take a closer look. Go ahead and make a new ticket with your findings, and if you know how to make a pull-request with patches I'll be happy to take a look at that too.

nicxvan, 2019-01-20:
I do know how to do a PR but I don't know where to look here to fix this. I'll open a couple of issues for things I've noticed and I'm willing to test.

johnboiles, 2019-02-02:
I'm getting the same thing on a raspberry pi using a docker image I built master as of right now (0ff11d858da9380e78246d9b0179d96c1d68d60d)

`/usr/local/bin/entrypoint.sh: line 48: can't create /etc/nut/ups.conf: Interrupted system call`

johnboiles, 2019-02-02:
Seems like it might be something about the use of EOF in the `entrypoint.sh` bash script, though I'm struggling to understand what it could be. Mounting my config as a volume in `/etc/nut/local` works just fine.

instantlinux, 2019-02-05:
@johnboiles Can you provide more detail: the file(s) you've mounted as /etc/nut/local, and the docker-compose.yml or _docker run_ command line options?

johnboiles, 2019-02-10:
I'm just mounting `ups.conf` but I still see `Interrupted system call` errors sometimes for other files. But they seem maybe less frequent? Here are my changes (for running on a pi, and also to mount the local ups.conf)

```diff
diff --git a/images/nut-upsd/Dockerfile b/images/nut-upsd/Dockerfile
index c144a32..339c794 100644
--- a/images/nut-upsd/Dockerfile
+++ b/images/nut-upsd/Dockerfile
@@ -1,4 +1,4 @@
-FROM alpine:3.8
+FROM arm32v6/alpine:3.8
 MAINTAINER Rich Braun "docker@instantlinux.net"
 ARG BUILD_DATE
 ARG VCS_REF
diff --git a/images/nut-upsd/docker-compose.yml b/images/nut-upsd/docker-compose.yml
index 8252d41..24b716f 100644
--- a/images/nut-upsd/docker-compose.yml
+++ b/images/nut-upsd/docker-compose.yml
@@ -11,6 +11,8 @@ services:
     secrets:
     - nut-upsd-password
     privileged: true
+    volumes:
+    - ${PWD}/conf:/etc/nut/local

 secrets:
   nut-upsd-password:
```

mtitus83, 2019-03-01:
Ran into a similar issue myself, google search landed me here but I was able to correct some mistakes after reviewing the docker-compose file posted by @nicxvan.  In my docker log I do see it still throws the "interruped system call" error however things look kosher and so far I have not had any issues.  Here's the my docker-compose.yml and logfile if any of this information is useful, I'm using the latest.  Just a side note I'm not using swarm currently so you'll note I'm not using the 'secrets' option and doing something unconventional to send the password through to the container at the moment since its just a lab environment.

`version: "3.1"

services:
  nut:
    environment:
      DRIVER: "usbhid-ups"
      API_USER: "upsmon"
      SERIAL: "CTHHW2001336"
      SECRET: nut-upsd-password
      PORT: /dev/usb/hiddev0
      USER: nut
      GROUP: nut
      NAME: CP1500AVRLCD
    image: instantlinux/nut-upsd
    container_name: nut
    hostname: nut
    domainname: hoodnet.net
    restart: always
    devices:
      - /dev/usb/hiddev0:/dev/usb/hiddev0
    volumes:
       - ./nut-upsd-password:/run/secrets/nut-upsd-password
    networks:
      vlan50:
        ipv4_address: x.x.x.x
      vlan70:
        ipv4_address: x.x.x.x
    privileged: true

networks:
  vlan50:
    external: true 
  vlan70:
    external: true 
`

`2019-03-01T03:28:22.935358464Z /usr/local/bin/entrypoint.sh: line 12: can't create /etc/nut/ups.conf: Interrupted system call
2019-03-01T03:28:25.651046653Z /usr/local/bin/entrypoint.sh: line 12: can't create /etc/nut/ups.conf: Interrupted system call
2019-03-01T03:28:28.663347821Z Network UPS Tools - UPS driver controller 3.8.0-3220-g866b921892
2019-03-01T03:28:28.663860363Z Network UPS Tools - Generic HID driver 0.41 (3.8.0-3220-g866b921892)
2019-03-01T03:28:28.756041222Z Using subdriver: CyberPower HID 0.4
2019-03-01T03:28:28.756066931Z USB communication driver 0.33
2019-03-01T03:28:28.805008727Z Network UPS Tools upsd 3.8.0-3220-g866b921892
2019-03-01T03:28:28.805039056Z Ignoring invalid pid number 0
2019-03-01T03:28:28.805121326Z listening on 0.0.0.0 port 3493
2019-03-01T03:28:28.805335351Z Connected to UPS [CP1500AVRLCD]: usbhid-ups-CP1500AVRLCD
2019-03-01T03:28:28.807119340Z Network UPS Tools upsmon 3.8.0-3220-g866b921892
2019-03-01T03:28:28.807144224Z    0.000000      Ignoring invalid pid number 0
2019-03-01T03:28:28.807222540Z    0.000099      Using power down flag file /etc/killpower
2019-03-01T03:28:28.807356331Z    0.000174      UPS: CP1500AVRLCD@localhost (master) (power value 1)
2019-03-01T03:28:28.807372748Z    0.000208      debug level is '1'
2019-03-01T03:28:28.809744905Z    0.002591      Trying to connect to UPS [CP1500AVRLCD@localhost]
2019-03-01T03:28:28.810031039Z    0.002897      Logged into UPS CP1500AVRLCD@localhost
`