instantlinux/docker-tools #9: error parsing [/etc/haproxy.cfg:27] : 'uri' needs an URI prefix

novafiddler, 2019-01-01:
Due to the entrypoint.sh script (it seems), the haproxy.cfg is generated with a blank 'uri' at the very bottom of haproxy.cfg everytime the machine is restarted. Creating a correct haproxy.cfg and setting it read-only does not work. This, for me, renders the entire machine unusable.

Also, the defaults in haproxy.cfg are not secure. Perhaps it may be best to create an environment variable that lets you define your own /etc/haproxy.cfg?

Thanks!

instantlinux, 2019-01-14:
I've addressed this in the 1.9.1-alpine-2.0.11-r0 version pushed to [instantlinux/haproxy-keepalived](https://hub.docker.com/r/instantlinux/haproxy-keepalived) today. Thanks for your suggestions, let me know if it's now able to meet your needs.
-rich

instantlinux, 2019-03-15:
Closing inactive issue, I'll take another look if you file a new issue or reopen.

closed 2019-03-15 by instantlinux