instantlinux/docker-tools #12: nut-upsd	- Question

Dinth, 2019-01-08:
I do apologize as this is not an issue per see, but i was wondering if nut-upsd will allow me to use multiple devices with multiple different drivers simultaneously using same docker instance?

instantlinux, 2019-01-08:
It's designed to auto-generate the configs for a single device and serial number. I've generally run multiple copies of this image, mapped to different ports, if I wanted to monitor more than one UPS device attached to a host. You could try adding a read-only volume mount of the /etc files to provide your own configs, and if that doesn't work with the current entrypoint script I'm happy to review if you'd like to submit a PR of a modified entrypoint script that supports a mounted config.

instantlinux, 2019-01-14:
Added this note to the README:

> If you have more than one UPS connected to a host, run more than one copy of this container and bind the container port 3493 from each to a separate TCP port.

It's now posted to [instantlinux/nut-upsd](https://hub.docker.com/r/instantlinux/nut-upsd).

closed 2019-01-14 by instantlinux