instantlinux/docker-tools #13: NUT UPSD container won't remain started - it says that the hid usb device can't be found.

nicxvan, 2019-01-19:
**What I'm trying to do**
Get NUT UPSD running

**The problem I'm having**
The container won't stay running.

**Error messages/logs**
Network UPS Tools - UPS driver controller 3.8.0-3220-g866b921892,
Network UPS Tools - Generic HID driver 0.41 (3.8.0-3220-g866b921892),
USB communication driver 0.33,
No matching HID UPS found,
Driver failed to start (exit status=1),

**O/S environment**
 - Distro: ubuntu 18.04
 - Docker version: Docker version 18.06.0-ce-dev, build 3dfd20c
 - Orchestrator (k8s, swarm, none): none, docker-compose

**Additional context**
Add any other context about the question here
I do see a /dev/usb/hiddev0 on the host machine. I set up a devices stanza and restarted it as well.
```
  nutups:
    container_name: nut_ups
    image: instantlinux/nut-upsd:2.7.4-r6
    environment:
      SERIAL: 2834BY0SM886900731
    network_mode: bridge
    ports:
      - 3493:3493
    devices:
      - /dev/usb/hiddev0:/dev/usb/hiddev0
    privileged: true
    secrets:
      - nut-upsd-password
```

nicxvan, 2019-01-19:
I have a TRIPP Lite 1500LCD which is connected by USB to the host machine.

instantlinux, 2019-01-20:
Are you able to get the nut package working outside docker? I see there are compatibility issues with the newer Tripp Lite models: see [issue #577](https://github.com/networkupstools/nut/issues/577) in the networkupstools upstream repo; check the links there for any insights you might get. For older models, there have been permissions issues (the image here runs process upsd as user _nut_): you could try using _docker exec_ into the container, run _ps_ to find the pid of upsd, kill it and restart with _/usr/sbin/upsd -u root_. If that turns out to be the issue, let me know and I can make a username parameter.

nicxvan, 2019-01-20:
I got a lot closer, let me highlight the changes I made here:

```  nutups:
    container_name: nut_ups
    restart: unless-stopped
    build:
      context: ./containers/nutupsd
    network_mode: bridge
    ports:
      - 3493:3493
    devices:
      - /dev/usb/hiddev0:/dev/usb/hiddev0
    privileged: true
    secrets:
      - nut-upsd-password
secrets:
  nut-upsd-password:
    file: ./nutpass```

Dockerfile I added:
```
COPY ups.conf /etc/nut/ups.conf
```
Right before the entrypoint copy

entrypoint changes, I deleted the cat to ups.conf
Then the ups.conf I copied is below.
```
[ups]
    driver = usbhid-ups
    port = auto
    vendorid = 09ae
```

After building and starting this I get the following logs:
```

 470.063090	Communications with UPS ups@localhost lost,
sh: wall: not found,
   0.003085	Logged into UPS ups@localhost,
 470.063034	Poll UPS [ups@localhost] failed - Data stale,
   0.000226	UPS: ups@localhost (master) (power value 1),
   0.000264	debug level is '1',
   0.002670	Trying to connect to UPS [ups@localhost],
   0.000125	Using power down flag file /etc/killpower,
   0.000000	fopen /var/run/upsmon.pid: No such file or directory,
Network UPS Tools upsmon 3.8.0-3220-g866b921892,
listening on 0.0.0.0 port 3493,
Connected to UPS [ups]: usbhid-ups-ups,
fopen /var/run/nut/upsd.pid: No such file or directory,
Network UPS Tools upsd 3.8.0-3220-g866b921892,
USB communication driver 0.33,
Network UPS Tools - Generic HID driver 0.41 (3.8.0-3220-g866b921892),
Using subdriver: TrippLite HID 0.82,
Network UPS Tools - UPS driver controller 3.8.0-3220-g866b921892```

nicxvan, 2019-01-20:
I also changed /dev/usb/hiddev0 to 660 and root:dialout

nicxvan, 2019-01-20:
I took a look at that issue you posted, it reminded me that I also added this:

```ACTION=="add", SUBSYSTEM=="usb", ATTRS{idVendor}==“09ae”, ATTRS{idProduct}==“2012”, MODE="0660", GROUP="nut"
```
To /etc/udev/rules.d/99-usb-serial.rules

instantlinux, 2019-01-20:
Watch for the new image to show up at Dockerhub, it will have the VENDORID parameter added. The [README](https://github.com/instantlinux/docker-tools/blob/master/images/nut-upsd/README.md) file here now has the information you've shared about Tripp Lite, thanks!

nicxvan, 2019-01-20:
No problem, let me close this.

closed 2019-01-20 by nicxvan