instantlinux/docker-tools #18: Postfix container - sasl secret

psyciknz, 2019-02-14:


psyciknz, 2019-02-14:
Having a few issues getting this container to work.

Should I be making a 
```
secrets:
  postfix-sasl-passwd:
    file: ./postfix/smtp_sasl_password_map
```
entry in the docker compose file?

where teh content is:
smtp.gmail.com:587      admin@mydomain.com:mypassword

ie, the format of a regular smtl_sasl_password_map?  As the error I'm getting is saying authentication needed.

Also not sure I get the link to the generic and virtusertable these end up in the etc/postfix/postfix.d folder, but it's referenced in /etc/postfix - they never seem to end up ther

instantlinux, 2019-02-22:
Confirming: the postfix-sasl-passwd secret should contain map entries as you suggest.
Also confirming: when you mount a volume into /etc/postfix/postfix.d, each file suffixed _.cf_ will be installed using _postconf_ and each file suffixed _.map_ will be copied into /etc/postfix (without the .map suffix) and then its corresponding _.db_ will be generated using _postmap_. For my own configuration, I have a master.cf, postfix.cf, transport.map and virtusertable.map in addition to aliases (which needs no suffix).

I think all you need to do is rename virtusertable to virtusertable.map and it'll get copied into /etc/postfix/virtusertable with a corresponding .db file. This container doesn't do anything with _generic_ which only contains documentation.

instantlinux, 2020-03-15:
Inactive issue: postfix has been updated a few times since.

closed 2020-03-15 by instantlinux