instantlinux/docker-tools #27: samba-dc: error when compose

vngvng, 2019-11-20:
**When compose the container, here are the error msg.**

app_1  | /usr/sbin/samba_dnsupdate: ERROR(runtime): uncaught exception - (9711, 'WERR_DNS_ERROR_RECORD_ALREADY_EXISTS')
app_1  | /usr/sbin/samba_dnsupdate:   File "/usr/lib/python2.7/site-packages/samba/netcmd/__init__.py", line 177, in _run
app_1  | /usr/sbin/samba_dnsupdate:     return self.run(*args, **kwargs)
app_1  | /usr/sbin/samba_dnsupdate:   File "/usr/lib/python2.7/site-packages/samba/netcmd/dns.py", line 940, in run

**And here are the log from container console.**
/usr/sbin/samba_dnsupdate: ERROR(runtime): uncaught exception - (9711, 'WERR_DNS_ERROR_RECORD_ALREADY_EXISTS')

/usr/sbin/samba_dnsupdate:   File "/usr/lib/python2.7/site-packages/samba/netcmd/__init__.py", line 177, in _run

/usr/sbin/samba_dnsupdate:     return self.run(*args, **kwargs)

/usr/sbin/samba_dnsupdate:   File "/usr/lib/python2.7/site-packages/samba/netcmd/dns.py", line 940, in run

/usr/sbin/samba_dnsupdate:     raise e

../source4/dsdb/dns/dns_update.c:330: Failed DNS update - with error code 29

instantlinux, 2019-12-22:
Samba maintainers are working in this area of code and it looks like they're not quite done fixing things. I"m tracking [bug #14209](https://bugzilla.samba.org/show_bug.cgi?id=14209) and expect to update the image once this gets released. A source code patch was uploaded yesterday so stay tuned.

instantlinux, 2020-01-06:
That bug fix is implemented in the upstream and I have published instantlinux/samba-dc:4.11.4-r1 (which swaps out python2.7 for python3). Please try that version and let me know if you still see this error.

instantlinux, 2020-02-26:
Closing after inactivity: problem should be fixed.

closed 2020-02-26 by instantlinux