instantlinux/docker-tools #29: -bash: sekret: command not found

eleaner, 2020-02-14:
Hi,

I am trying to install mariadb(only) using your image and existing etcd cluster, and I am failing miserably
"Create a random root password:" section contains line:
`sekret enc /dev/shm/new.yaml >secrets/$SECRET`

I am using ubuntu 16.04 and despite rampaging google I cannot find such tool

Could you please help?

eleaner, 2020-02-14:
ok, I think you meant this tool
https://github.com/nownabe/sekret

but I am trying to use docker swarm, not kubernetes
do I still have to follow this procedure?

instantlinux, 2020-02-15:
You can use the [docker-compose.yml](https://github.com/instantlinux/docker-tools/blob/master/images/mariadb-galera/docker-compose.yml) file here in this repo for a now-deprecated example of how to get it working with swarm. You can generate a secret and store it in swarm using the [docker secret create](https://docs.docker.com/engine/reference/commandline/secret_create/) command. Thanks for bringing this to my attention, will update the README for mariadb-galera to make it more clear.

eleaner, 2020-02-15:
thank you @instantlinux 

I am using exactly this compose file modified to also define secrets
yet the cluster is failing to bootstrap complaining about the root login.
I could post here part of logs but I am not sure if you'd be interested in helping me fix swarm issues. 
I assume you moved to Kubernetes

instantlinux, 2020-02-17:
Yes, it's been a long time since swarm. But your issue may not be with swarm itself: I'll keep this issue open for a while: other swarm users might benefit from whatever you come up with. What's the error? You should have a secret _mysql-root-password_ that gets set (as file /run/secrets/mysql-root-password) at initial start. The container also supports a _sst-auth-password_ secret.

instantlinux, 2020-02-26:
Updated [README.md](https://github.com/instantlinux/docker-tools/blob/master/images/mariadb-galera/README.md) with clarification that either sops or sekret can be used to maintain the secrets files (these days my preference is sops, see the [k8s](https://github.com/instantlinux/docker-tools/blob/master/k8s) Makefiles). Also clarified that the instructions are for kubernetes, with a link to the legacy docker-compose.yml.

closed 2020-02-26 by instantlinux