instantlinux/docker-tools #30: weewx: volumes bind for persistent data

pbartko, 2020-02-18:
I am using your latest weewx docker image from Dockerhub but have run into a problem trying to bind volumes for persistent data.  I was successful in binding /secrets folder for passwords to my host system. However when I bind the /home/weewx folder to my host system, the docker won't run.  I've attached the log file here:
[weewx log2.txt](https://github.com/instantlinux/docker-tools/files/4217098/weewx.log2.txt)

Any help is much appreciated.

Thanks,

Pete

instantlinux, 2020-04-17:
The /home/weewx folder has the application code in it. You could mount specific files within it if you want to add or override those in the container, but the code needs to be available in order to run.

instantlinux, 2020-05-04:
Please reopen if you have further questions.

closed 2020-05-04 by instantlinux