instantlinux/docker-tools #32: build all tools with alpine for arm32v6 and arm64v8 too

c64emulator, 2020-02-20:
With arm32v6/alpine and arm64v8/alpine and using manifest to build multiarch images all images will have similiar footprint and you need only one Dockerfile for all.

instantlinux, 2020-02-26:
Can you explain further? The nut-upsd arm image is built using balenalib/raspberry-pi-alpine:3.9 using a separate Dockerfile (so it's likely to fall out of date without giving it manual attention). If you want to submit a PR or add a doc here for making this happen, I'm interested to hear more.

instantlinux, 2020-05-05:
The doc isn't particularly good but [this blog post](https://digilution.io/posts/multiarch-docker-builds/) kinda gives me a hint. Looks like I should be able to use my existing Dockerfiles for each image (at least, the ones that use alpine:3.11 or others with similar multi-platform support), and change the hooks/build files to add the `buildx` and `--platform` CLI options (assuming hub.docker.com's build service supports buildx; in any case I've been thinking of moving the builds over to gitlab's CI service and mirroring them to hub.docker.com). Let me know if you have examples of this idea.

instantlinux, 2020-05-07:
Did an experiment on the _hub.docker.com_ build service: they've gotten so far out of date that this will have to be implemented after I've moved off this build service (to gitlab). The _buildx_ command is unsupported; here's the ancient version they're running (output taken from a build like [this one](https://hub.docker.com/repository/registry-1.docker.io/instantlinux/rsyslogd/builds/12bd23e8-ef02-4fc5-8faa-cd5dc7a2202a):
```
+ docker version
Client:
Version: 17.06.2-ee-6
API version: 1.30
Go version: go1.8.3
Git commit: e75fdb8
Built: Mon Nov 27 22:45:33 2017
OS/Arch: linux/amd64
Server:
Version: 18.03.1-ee-3
API version: 1.37 (minimum version 1.12)
Go version: go1.10.2
Git commit: b9a5c95
Built: Thu Aug 30 18:42:30 2018
OS/Arch: linux/amd64
Experimental: false
```
Once I set CI up on a system that supports the following script, I can move forward:
```
docker buildx create --name multi
docker buildx use multi
docker buildx build -t $IMAGE_NAME . \
  --build-arg VCS_REF=$(git rev-parse HEAD) \
  --build-arg BUILD_DATE=$(date -u +%Y-%m-%dT%H:%M:%SZ)
```

instantlinux, 2020-06-07:
Success - I have moved builds for the following images from dockerhub to gitlab.com, as multi-arch builds for amd64, arm64, arm/v6 and arm/v7. See the [.gitlab-ci.yml](https://gitlab.com/instantlinux/docker-tools/-/blob/master/.gitlab-ci.yml) with child pipeline [.image-gitlab-ci.yml](https://gitlab.com/instantlinux/docker-tools/-/blob/master/.image-gitlab-ci.yml) and [Makefile.docker-image](https://gitlab.com/instantlinux/docker-tools/-/blob/master/lib/build/Makefile.docker_image) for how it's put together.
  - data-sync
  - dhcpd-dns-pxe
  - dovecot
  - ez-ipupdate
  - git-dump
  - git-pull
  - postfix
  - proftpd
  - python-builder
  - rsyslogd
  - samba
  - vsftpd

instantlinux, 2020-06-17:
Remaining images are now published as multi-arch arm/v7, arm/v6, arm64 and amd64:

* blacklist
* mt-daapd
* nagiosql
* nut-upsd
* udp-nginx-proxy
* postfix-python

I have three others that have fallen out of date, and created separate tickets to address them (mariadb and weewx). The samba-dc images are still built for ARM in separate repos because the multi-arch build runs too slowly under qemu.

Thanks for inspiring me to get this work done!

closed 2020-06-17 by instantlinux