instantlinux/docker-tools #34: nut-upsd not listening to 0.0.0.0 if `/etc/nut/local/ups.conf` is provided

ugomeda, 2020-04-30:
When launching upsd with a custom `ups.conf` configuration, the line `LISTEN 0.0.0.0` is not appended to `/etc/nut/upsd.conf`.

This is my docker-compose configuration :

```
  networkupstools:
    image: instantlinux/nut-upsd
    privileged: true
    restart: always
    mem_limit: 128m
    cpus: 1
    dns: 192.168.101.1
    devices:
      - /dev/bus/usb:/dev/bus/usb
    volumes:
      - ./ups.conf:/etc/nut/local/ups.conf:ro
      - ./secrets/nut-upsd-password:/run/secrets/nut-upsd-password:ro
    networks:
      internal:
        ipv4_address: 192.168.101.7
```

I think the cause is a typo in `docker-tools/images/nut-upsd/entrypoint.sh` (lines 26-32) :

```
  if [ -e /etc/nut/local/ups.conf ]; then
    cp /etc/nut/local/ups.conf /etc/nut/ups.conf
  else
    cat <<EOF >>/etc/nut/upsd.conf
LISTEN 0.0.0.0
EOF
  fi
```

It should check for the presence of `/etc/nut/local/upsd.conf`, not `/etc/nut/local/ups.conf` :

```
  if [ -e /etc/nut/local/upsd.conf ]; then
    cp /etc/nut/local/upsd.conf /etc/nut/upsd.conf
  else
    cat <<EOF >>/etc/nut/upsd.conf
LISTEN 0.0.0.0
EOF
  fi
```

instantlinux, 2020-05-05:
Thanks for reporting this, I had an earlier report of the same issue and have pushed a fix: https://github.com/instantlinux/docker-tools/pull/23.

ugomeda, 2020-05-05:
Amazing !

I just tested it and it works properly now, thank you !

closed 2020-05-05 by ugomeda