instantlinux/docker-tools #36: Bugs in Docker image for NagiosQL, just FYI

MichaelTiernan, 2020-05-10:
Just as an FYI, the Docker image doesn't seem to be complete:

~~~sh
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/nagios/bin
+ NAGIOS_ETC=/opt/nagios/etc
+ NAGIOS_OBJ=/opt/nagios/etc/objects
+ [ ! -f /etc/timezone ]
+ [ -e /run/secrets/nagiosql-db-password ]
+ sed -e s/{{ DB_HOST }}/db00/ -e s/{{ DB_NAME }}/nagiosql/ -e s/{{ DB_PASS }}// -e s/{{ DB_PORT }}/3306/ -e s/{{ DB_USER }}/nagiosql/ /var/www/nagiosql/config/settings.php.j2
+ sed -i -e s:use_timezone=UTC:use_timezone=UTC: /opt/nagios/etc/nagios.cfg
sed: can't read /opt/nagios/etc/nagios.cfg: No such file or directory
~~~

and the Dockerfile as provided on docker hub seems to have a couple of problems too:

~~~sh
docker build .
[...]
Step 13/21 : COPY src /tmp/
COPY failed: stat /var/lib/docker/tmp/docker-builder040151770/src: no such file or directory
~~~

If that step is removed by commenting it out:

~~~sh
Step 14/20 : RUN 	sed -i -e 's/;date.timezone =/date.timezone = UTC/' 		/etc/php/7.2/apache2/php.ini && cd /tmp && 	curl -sLo $NAGIOSQL_DOWNLOAD 		https://sourceforge.net/projects/nagiosql/files/nagiosql/NagiosQL%20${NAGIOSQL_VERSION}/${NAGIOSQL_DOWNLOAD} && 	echo "$NAGIOSQL_SHA	 $NAGIOSQL_DOWNLOAD" | sha256sum -c && mkdir /var/www/nagiosql && 	tar xjf $NAGIOSQL_DOWNLOAD -C /var/www/nagiosql --strip-components=1 && 	groupadd -g $NAGIOS_GID nagios && 	useradd -u $NAGIOS_UID -c Nagios -s /sbin/nologin -g nagios nagios && 	usermod -G nagios $APACHE_USER && 	mv /tmp/nagiosql.conf /etc/apache2/conf-enabled/ && 	mv /tmp/settings.php.j2 /var/www/nagiosql/config/ && 	rm -fr /tmp/* /var/log/[b-w]*
 ---> Running in a6457adbaa89
nagiosql-3.4.0-git2019-04-26.tar.bz2: OK

mv: cannot stat '/tmp/nagiosql.conf': No such file or directory

The command '/bin/sh -c sed -i -e 's/;date.timezone =/date.timezone = UTC/' 		/etc/php/7.2/apache2/php.ini && cd /tmp && 	curl -sLo $NAGIOSQL_DOWNLOAD 		https://sourceforge.net/projects/nagiosql/files/nagiosql/NagiosQL%20${NAGIOSQL_VERSION}/${NAGIOSQL_DOWNLOAD} && 	echo "$NAGIOSQL_SHA	 $NAGIOSQL_DOWNLOAD" | sha256sum -c && mkdir /var/www/nagiosql && 	tar xjf $NAGIOSQL_DOWNLOAD -C /var/www/nagiosql --strip-components=1 && 	groupadd -g $NAGIOS_GID nagios && 	useradd -u $NAGIOS_UID -c Nagios -s /sbin/nologin -g nagios nagios && 	usermod -G nagios $APACHE_USER && 	mv /tmp/nagiosql.conf /etc/apache2/conf-enabled/ && 	mv /tmp/settings.php.j2 /var/www/nagiosql/config/ && 	rm -fr /tmp/* /var/log/[b-w]*' returned a non-zero code: 1
~~~

While a lot of this may be "Left as an exercise for the reader", IMNSHO, some mention of the prerequirements might be in order.

With all that said, thank you *very* much for your efforts toward helping us with this and for providing this solution to begin with. :)

instantlinux, 2020-05-13:
Thanks for the report, I'll take a look and see what clarifications / fixes make sense.

instantlinux, 2020-05-22:
The README is updated and a new version 3.4.1 of the image is available. It clarifies, with a better error message, that the jasonrivers/nagios image must be set up first, with the same volume for /opt/nagios/etc as nagiosql. The docker-compose.yml does this for you.

instantlinux, 2020-06-17:
I've rebuilt the image under alpine so it now supports raspberry pi as well as x86 platforms, and further improved the entrypoint script and Dockerfile. Results are now posted to the 3.4.1 tag at https://hub.docker.com/repository/docker/instantlinux/nagiosql.

closed 2020-06-17 by instantlinux