instantlinux/docker-tools #37: Where to start to get ox app working?

jiripaskov, 2020-05-20:
Hello I would like to run your ox app suite docker but I´m a newbie. I  downloaded k8s from git using git clone https://github.com/instantlinux/docker-tools.git but now I´m totally lost. I can see there is docker-compose.yml in "images" directory but I´m not sure if I shoudl work with it... PLease oculd you point me how to prepare things to run ox app docker? Thank you very much.

instantlinux, 2020-06-24:
Starting to look at this now, I'll update the README and the docker-compose / kubernetes.yaml files with any improvements I can make. The installer does take a fair amount of time, and throws a lot of expected errors during setup -- I'll have to document those and get the installation to run smoothly again now that several release updates have happened since my initial success.

instantlinux, 2020-06-24:
Update: I haven't been able to get 7.10.3-x running at all, and there have been some breaking changes in the 7.10.2-x updates since this image last worked reliably.

Turns out to be more work than I expected, hopefully I can at least generate a working 7.10.2-x image. Thanks for the bug-report -- while your question is probably an elementary "how do I set up the variables to make docker-compose happy", turns out that there isn't a way to run the versions generated in the past few months. This is a very old, unsupported (at least, under Docker) piece of OSS from a company based in Germany. I'll do my best to get the docker installation working again.

instantlinux, 2020-06-26:
I'm unable to get the application working, either with latest 7.10.2 or 7.10.3 versions. The tagged image 7.10.1 might work but the more recent ones do not. And after two days of beating my head against this, I'm unable to fix the problems.

If any user out there is familiar with Open Xchange Appsuite or has support from the vendor, I'll be happy to accept pull-requests or suggestions on how to make it work. (I am running into connection problems between the running image and dovecot/imap due to outdated SSL versions; and the internal services aren't coming up the way they did in 7.10.1 - I just get the error "Failed to start application. Maybe you have connection problems" and no error logs explaining why.

Sorry to share this bad news.

instantlinux, 2020-06-26:
Submitted a bug #68833 to the vendor's bugzilla.

instantlinux, 2020-09-20:
I assume the product is no longer getting maintained: my bug report got no response for the past 90 days, and there's now a notice that the bug-tracking system itself will be retired in 60 days (see [decommissioning notice](https://forum.open-xchange.com/forum/open-xchange-app-suite/open-xchange-app-suite-announcements/93711-bugzilla-bug-reporting-changes#post93711)).

frankf-cgn, 2020-10-18:
I have 7.10.4 running in docker with slight modifications to the stuff in this repo. But the modifications should only concern my special requirements (e.g. imap authentification). 
Can you tell me, which problems you encounter? Sadly, I cannot access #68833 on bugzilla. Maybe I can help out and un-deprecate this part of the repo .

instantlinux, 2020-10-22:
@frankf-cgn Try the tag 7.10.4-9. I welcome pull-requests to fix issues you find (could be in Dockerfile or entrypoint.sh). If the image works for you, please add a note here about the environment you're using.

instantlinux, 2023-09-04:
Closing due to inactivity.

closed 2023-09-04 by instantlinux