instantlinux/docker-tools #39: samba-dc: multiple WERR_DNS_ERROR_RECORD_ALREADY_EXISTS

ichundu, 2020-06-22:
Hello,

I'm getting the following error multiple times when running `docker-compose up -d`.
This was referenced in this [issue](https://github.com/instantlinux/docker-tools/issues/27) also

```
/usr/sbin/samba_dnsupdate: ERROR(runtime): uncaught exception - (9711, 'WERR_DNS_ERROR_RECORD_ALREADY_EXISTS')
/usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/__init__.py", line 186, in _run
/usr/sbin/samba_dnsupdate:     return self.run(*args, **kwargs)
/usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/dns.py", line 945, in run
/usr/sbin/samba_dnsupdate:     raise e
/usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/dns.py", line 940, in run
/usr/sbin/samba_dnsupdate:     dns_conn.DnssrvUpdateRecord2(dnsserver.DNS_CLIENT_VERSION_LONGHORN,
```

Here is my docker-compose.yml:
```yaml
version: "3.1"

services:
  app:
    container_name: samba_dc
    image: ${REGISTRY_URI:-instantlinux}/samba-dc:latest
    restart: unless-stopped
    network_mode: host
    cap_add:
      - SYS_ADMIN
    hostname: ${SAMBADC_HOSTNAME:-dc.example.com}
    environment:
      DOMAIN_ACTION: ${SAMBADC_DOMAIN_ACTION:-join}
      INTERFACES: ${SAMBADC_INTERFACES:-lo eth0}
      REALM: ${SAMBA_REALM:-ad.example.com}
      TZ: ${TZ:-UTC}
      WORKGROUP: ${SAMBA_WORKGROUP:-WORKGROUP}
    volumes:
      - etc:/etc/samba
      - lib:/var/lib/samba
    ports:
      - 53:53
      - 53:53/udp
      - 88:88
      - 88:88/udp
      - 135:135
      - 137-138:137-138/udp
      - 139:139
      - 389:389
      - 389:389/udp
      - 445:445
      - 464:464
      - 464:464/udp
      - 636:636
      - 3268-3269:3268-3269
      - 49152-65535
    secrets:
      - samba-admin-password

volumes:
  etc:
  lib:
secrets:
  samba-admin-password:
    file: /etc/samba/samba-admin-password
```

instantlinux, 2020-06-22:
Thanks for your report. As part of another issue #32, I've added multi-architecture support to this image. And I noticed that the Alpine package maintainer for _bind-tools_ finally added the gssapi module required for samba_dnsupdate.

After doing a `docker pull instantlinux/samba-dc:latest` to get the image now built with the alpine:3.12 version of bind tols, try this command to see what's happening on your instance:
```
samba_dnsupdate --verbose --all-names
```
For those that show up as already-existing, try removing them using `samba-tool dns delete <name>` and see if the dnsupdate works afterward.

ichundu, 2020-06-22:
It's still failing:
https://pastebin.com/br6N1eDa

In the log there is this line:
```
Failed to get Kerberos credentials, falling back to samba-tool: kinit for AD$@AD.VAGRANT.AL failed (Cannot contact any KDC for requested realm)
```

This is the full log of samba domain provision on first start:
https://pastebin.com/JyAkB90j

I don't know if there's something wrong with my setup:

- Running on vagrant with ubuntu 20.04
- systemd-resolved service disabled
- my /etc/resolv.conf file on ubuntu host:
  ```
  # default gateway
  nameserver 192.168.121.1
  ```
- my /etc/hosts file on ubuntu host:
  ```
  127.0.0.1	localhost

  # The following lines are desirable for IPv6 capable hosts
  ::1     localhost ip6-localhost ip6-loopback
  ff02::1 ip6-allnodes
  ff02::2 ip6-allrouters

  127.0.0.1 ubuntu2004.localdomain
  192.168.121.17 node1.ad.vagrant.al node1
  ```

- docker-compose customizations via env:
  ```
  export SAMBADC_HOSTNAME=ad.vagrant.al
  export SAMBADC_DOMAIN_ACTION=provision
  export SAMBA_REALM=ad.vagrant.al
  export TZ=Europe/Tirane
  export SAMBA_WORKGROUP=vagrant
  ```

instantlinux, 2020-06-23:
I set up your docker-compose file, with an `env.sh` file containing your env settings, and third file `samba-admin-password` containing the value `foobar@123` and ran it on my ubuntu 18.04 host (not under vagrant). Here's what the output is for me:
```
$ docker logs -f samba_dc
Set timezone
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/main/x86_64/APKINDEX.tar.gz
fetch http://dl-cdn.alpinelinux.org/alpine/v3.12/community/x86_64/APKINDEX.tar.gz
(1/1) Installing tzdata (2020a-r0)
Executing busybox-1.31.1-r16.trigger
OK: 168 MiB in 110 packages
INFO 2020-06-23 04:06:32,298 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2133: Looking up IPv4 addresses
WARNING 2020-06-23 04:06:32,322 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2143: No IPv4 address will be assigned
INFO 2020-06-23 04:06:32,322 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2150: Looking up IPv6 addresses
WARNING 2020-06-23 04:06:32,324 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2157: No IPv6 address will be assigned
INFO 2020-06-23 04:06:33,191 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2301: Setting up share.ldb
INFO 2020-06-23 04:06:33,212 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2305: Setting up secrets.ldb
INFO 2020-06-23 04:06:33,230 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2311: Setting up the registry
INFO 2020-06-23 04:06:33,326 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2314: Setting up the privileges database
INFO 2020-06-23 04:06:33,387 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2317: Setting up idmap db
INFO 2020-06-23 04:06:33,408 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2324: Setting up SAM db
INFO 2020-06-23 04:06:33,412 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #897: Setting up sam.ldb partitions and settings
INFO 2020-06-23 04:06:33,414 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #909: Setting up sam.ldb rootDSE
INFO 2020-06-23 04:06:33,418 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1338: Pre-loading the Samba 4 and AD schema
Unable to determine the DomainSID, can not enforce uniqueness constraint on local domainSIDs

INFO 2020-06-23 04:06:33,466 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1416: Adding DomainDN: DC=ad,DC=vagrant,DC=al
INFO 2020-06-23 04:06:33,482 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1448: Adding configuration container
INFO 2020-06-23 04:06:33,497 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1463: Setting up sam.ldb schema
INFO 2020-06-23 04:06:40,108 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1481: Setting up sam.ldb configuration data
INFO 2020-06-23 04:06:40,445 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1522: Setting up display specifiers
INFO 2020-06-23 04:06:44,720 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1530: Modifying display specifiers and extended rights
INFO 2020-06-23 04:06:44,794 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1537: Adding users container
INFO 2020-06-23 04:06:44,797 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1543: Modifying users container
INFO 2020-06-23 04:06:44,799 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1546: Adding computers container
INFO 2020-06-23 04:06:44,802 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1552: Modifying computers container
INFO 2020-06-23 04:06:44,805 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1556: Setting up sam.ldb data
INFO 2020-06-23 04:06:45,113 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1586: Setting up well known security principals
INFO 2020-06-23 04:06:45,206 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1600: Setting up sam.ldb users and groups
INFO 2020-06-23 04:06:45,469 pid:17 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1608: Setting up self join
Repacking database from v1 to v2 format (first record CN=ms-DFS-Ttl-v2,CN=Schema,CN=Configuration,DC=ad,DC=vagrant,DC=al)
Repack: re-packed 10000 records so far
Repacking database from v1 to v2 format (first record CN=serversContainer-Display,CN=406,CN=DisplaySpecifiers,CN=Configuration,DC=ad,DC=vagrant,DC=al)
Repacking database from v1 to v2 format (first record CN=ipsecNFA{594272E2-071D-11D3-AD22-0060B0ECCA17},CN=IP Security,CN=System,DC=ad,DC=vagrant,DC=al)
INFO 2020-06-23 04:06:47,833 pid:17 /usr/lib/python3.8/site-packages/samba/provision/sambadns.py #1138: Adding DNS accounts
INFO 2020-06-23 04:06:47,866 pid:17 /usr/lib/python3.8/site-packages/samba/provision/sambadns.py #1172: Creating CN=MicrosoftDNS,CN=System,DC=ad,DC=vagrant,DC=al
INFO 2020-06-23 04:06:47,905 pid:17 /usr/lib/python3.8/site-packages/samba/provision/sambadns.py #1185: Creating DomainDnsZones and ForestDnsZones partitions
INFO 2020-06-23 04:06:47,977 pid:17 /usr/lib/python3.8/site-packages/samba/provision/sambadns.py #1190: Populating DomainDnsZones and ForestDnsZones partitions
Repacking database from v1 to v2 format (first record DC=i.root-servers.net,DC=RootDNSServers,CN=MicrosoftDNS,DC=DomainDnsZones,DC=ad,DC=vagrant,DC=al)
[... cut ...]
Doing a full scan on CN=Configuration,DC=ad,DC=vagrant,DC=al and looking for deleted objects
Doing a full scan on DC=ad,DC=vagrant,DC=al and looking for deleted objects
```
Looks identical up to the point of the line `TLS self-signed keys generated OK` which is then followed by the 4 `full scan` lines when it succeeds, or the DNS_ERROR message you've reported. 

Perhaps it needs to run under ubuntu 20.04 and/or under vagrant to reproduce the problem outside your environment? I can try another docker host where I have 20.04 set up.

instantlinux, 2020-06-23:
My existing ubuntu 20.04 hardware installation works fine. And here are the steps I did to reproduce under vagrant:
```
apt install vagrant
# edit new Vagrantfile to specify box bento/ubuntu-20.04
vagrant up
vagrant ssh
sudo su -
curl -fsSL https://download.docker.com/linux/ubuntu/gpg | sudo apt-key add -
apt update
apt install docker-ce
sudo curl -L "https://github.com/docker/compose/releases/download/1.26.0/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose
chmod +x /usr/local/bin/docker-compose
usermod -G docker -a vagrant
exit
```
Then reconnect and
```
echo -n foobar@123 >samba-admin-password
cat <<EOT >env.sh 
export SAMBADC_HOSTNAME=ad.vagrant.al
export SAMBADC_DOMAIN_ACTION=provision
export SAMBA_REALM=ad.vagrant.al
export TZ=Europe/Tirane
export SAMBA_WORKGROUP=vagrant
EOT
# same for docker-compose.yml as above
source env.sh
docker-compose up -d
docker logs -f samba_dc
```
It does generate these error messages as you report, but then it also successfully does this:
```
Doing a full scan on DC=ForestDnsZones,DC=ad,DC=vagrant,DC=al and looking for deleted objects
Doing a full scan on DC=DomainDnsZones,DC=ad,DC=vagrant,DC=al and looking for deleted objects
Doing a full scan on CN=Configuration,DC=ad,DC=vagrant,DC=al and looking for deleted objects
Doing a full scan on DC=ad,DC=vagrant,DC=al and looking for deleted objects
```
As near as I can tell, the problem you've reported is cosmetic and only reproducible under vagrant. After the above, I can see this query succeeds:
```
# samba-tool dns query localhost ad.vagrant.al @ ALL -U Administrator
Password for [VAGRANT\Administrator]:
  Name=, Records=3, Children=0
    SOA: serial=37, refresh=900, retry=600, expire=86400, minttl=3600, ns=ad.ad.vagrant.al., email=hostmaster.ad.vagrant.al. (flags=600000f0, serial=37, ttl=3600)
    NS: ad.ad.vagrant.al. (flags=600000f0, serial=1, ttl=900)
    A: 10.0.2.15 (flags=600000f0, serial=1, ttl=900)
  Name=_msdcs, Records=0, Children=0
  Name=_sites, Records=0, Children=1
  Name=_tcp, Records=0, Children=4
  Name=_udp, Records=0, Children=2
  Name=ad, Records=1, Children=0
    A: 10.0.2.15 (flags=f0, serial=1, ttl=900)
  Name=DomainDnsZones, Records=0, Children=2
  Name=ForestDnsZones, Records=0, Children=2
```

instantlinux, 2020-06-23:
Please escalate this to the upstream developers if this is blocking you. The team's bug-reporting site is:
https://bugzilla.samba.org/enter_bug.cgi

As near as I can tell, this is a new issue that has not been reported previously, and isn't related to the fix done about 6 months ago.

ichundu, 2020-06-23:
I will test with a different setup without vagrant when I have a bit more time, but if it's just cosmetic errors then it's fine.
Will post my findings here if I find something relevant.

Thanks for your time! Please close the issue.

instantlinux, 2020-06-24:
OK thanks for the input, and if you report this to the samba maintainers, you can link back to this ticket for the steps-to-reproduce that I posted here.

closed 2020-06-24 by instantlinux