instantlinux/docker-tools #40: kinit: Cannot find KDC for realm "DC.DGD.PW" while getting initial credentials

DylanDonkersgoed, 2020-07-18:
I'm going to preface here by saying this may be more of a Samba/AD issue than an issue with this docker container. I'm afraid I don't have too much experience with AD. So, sorry if this is in the wrong place.

My issue is that when I try to init a kerberos ticket (e.g. with `kinit Dylan`) I get the error in the title. What information I could find searching around indicates this is likely some kind of DNS error (or is related to some /etc/krb5.conf setting but these seem to be set correctly by the container as far as I can tell). However, dc.dgd.pw and ad.dgd.pw resolve correctly (to the same host) when pinged from within the container. I've tested this both from within and outside the container (with the krb5.conf file copied to the appropriate host) with the same result.

I _do_ see some errors related to DNS records already existing when samba starts up. However, based on https://github.com/instantlinux/docker-tools/issues/39 it looks like those are only cosmetic.

DNS is actually configured on an OpenWRT based router. I know that it is recommended to configure it in AD, but from what I can tell this is so you can more easily perform DNS updates/add additional domains within AD which is not a likely scenario for me.

Here's the log I see when samba is started up:

> samba_1  | samba version 4.12.2 started.
samba_1  | Copyright Andrew Tridgell and the Samba Team 1992-2020
samba_1  | binary_smbd_main: samba: using 'standard' process model
samba_1  | /usr/sbin/winbindd: winbindd version 4.12.2 started.
samba_1  | /usr/sbin/winbindd: Copyright Andrew Tridgell and the Samba Team 1992-2020
samba_1  | /usr/sbin/smbd: smbd version 4.12.2 started.
samba_1  | /usr/sbin/smbd: Copyright Andrew Tridgell and the Samba Team 1992-2020
samba_1  | /usr/sbin/smbd: INFO: Profiling support unavailable in this build.
samba_1  | /usr/sbin/winbindd: initialize_winbindd_cache: clearing cache and re-creating with version number 2
samba_1  | /usr/sbin/winbindd: daemon_ready: daemon 'winbindd' finished starting up and ready to serve connections
samba_1  | /usr/sbin/smbd: daemon_ready: daemon 'smbd' finished starting up and ready to serve connections
samba_1  | /usr/sbin/smbd: Failed to fetch record!
samba_1  | /usr/sbin/samba_dnsupdate: ERROR(runtime): uncaught exception - (9711, 'WERR_DNS_ERROR_RECORD_ALREADY_EXISTS')
samba_1  | /usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/__init__.py", line 186, in _run
samba_1  | /usr/sbin/samba_dnsupdate:     return self.run(*args, **kwargs)
samba_1  | /usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/dns.py", line 945, in run
samba_1  | /usr/sbin/samba_dnsupdate:     raise e
samba_1  | /usr/sbin/samba_dnsupdate:   File "/usr/lib/python3.8/site-packages/samba/netcmd/dns.py", line 940, in run
samba_1  | /usr/sbin/samba_dnsupdate:     dns_conn.DnssrvUpdateRecord2(dnsserver.DNS_CLIENT_VERSION_LONGHORN,
samba_1  | /usr/sbin/samba_dnsupdate: <REPEATED MANY TIMES>
samba_1  | /usr/sbin/samba_dnsupdate: 
samba_1  | dnsupdate_nameupdate_done: Failed DNS update with exit code 37
samba_1  | Doing a full scan on DC=ForestDnsZones,DC=ad,DC=dgd,DC=pw and looking for deleted objects
samba_1  | Doing a full scan on DC=DomainDnsZones,DC=ad,DC=dgd,DC=pw and looking for deleted objects
samba_1  | Doing a full scan on CN=Configuration,DC=ad,DC=dgd,DC=pw and looking for deleted objects
samba_1  | Doing a full scan on DC=ad,DC=dgd,DC=pw and looking for deleted objects

And here's my configuration:

```yaml
version: "3.1"

services:
  samba:
    image: instantlinux/samba-dc:latest
    restart: always
    network_mode: host
    hostname: dc.dgd.pw
    environment:
      INTERFACES: eno1
      REALM: ad.dgd.pw
      TZ: America/Toronto
      WORKGROUP: WORKGROUP
    volumes:
      - /opt/samba/etc/conf.d:/etc/samba/conf.d/
      - /opt/samba/lib:/var/lib/samba
      - /mnt/files:/mnt/files
    ports:
      - 53:53
      - 53:53/udp
      - 88:88
      - 88:88/udp
      - 135:135
      - 137-138:137-138/udp
      - 139:139
      - 389:389
      - 389:389/udp
      - 445:445
      - 464:464
      - 464:464/udp
      - 636:636
      - 3268-3269:3268-3269
      - 49152-65535
    secrets:
      - samba-admin-password
    privileged: true

volumes:
  etc:
  lib:
secrets:
  # Must manually create file with secret at this location
  samba-admin-password:
    file: /opt/samba/samba-admin-password
```

I also have an extension to the samba configuration in conf.d/100-shares.conf:

```
[global]
        idmap_ldb:use rfc2307 = Yes
        map to guest = Bad User

[public]
        path = /mnt/files
        read only = no
        guest ok = yes
        guest only = yes
```

And here's the krb5.conf file that's generated:

```
[libdefaults]
        default_realm = AD.DGD.PW
        dns_lookup_realm = false
        dns_lookup_kdc = true

[realms]
AD.DGD.PW = {
        default_domain = ad.dgd.pw
}

[domain_realm]
        dc = AD.DGD.PW
```

psaxton, 2020-08-24:
This is an AD setup issue. The DNS server run by samba-ad-dc includes special SRV records for the machines to find the appropriate resources.

In openWRT, you can send just the DNS requests for dc.dgd.pw and ad.dgd.pw to your AD server in the Network -> DHCP -> DNS Forwarding field in the format "/dc.dgd.pw/(host IP)". This allows you to continue using openWRT for your DHCP and primary DNS services.

instantlinux, 2023-09-04:
Closing due to inactivity.

closed 2023-09-04 by instantlinux