instantlinux/docker-tools #41: not that ecuret battery.charge will wall power is disconnected

eshchar, 2020-07-30:
when the power from the wall is disconnected and i am on the battery the percentage is displayed as so:
from 100 to 80 and then it jump to 30 and continue to drop (the UPS is NEW).
i now that the percentage is reported by the UPS.

but you can calculate the percentage According to the Voltage :
[image]

sow the calculation cod be (in my case i have a 4cells in line sow 4 * 12 = 48) :
if i use the picture info about battery then the 100% is 50.92V and the minimum recommended is 46.04V
then 50.92 - 46.04 = 4.88 => 4.88/100 = 0.0488 so every drop by 0.0488 volts is 1%.
i now that the the ups isn't that accurate so you can say that every 0.1V is 2%.

-so can you please give an option to choose my option (with a disclaimer) to show the percentage.
-and please add to the info displayed the battery Voltage.

psaxton, 2020-08-24:
Please include the image you are having difficulty with in your problem description. I assume you are using the NUTUPS image. This image is just a docker container of the NUTUPS application. Please consider filing a feature request/defect with the application project: https://github.com/networkupstools/nut

Also, check out their very detailed project site including the manual and a number of how-to's: https://networkupstools.org/index.html

eshchar, 2020-08-24:
my mistake.

closed 2020-08-24 by eshchar