instantlinux/docker-tools #44: samba-dc does not set a dns forwarder

dafzor, 2020-08-23:
DNS is very important for active directory.

samba-dc should have an additional environment variable to set DNS forwarder making it easier to USE it as a main DNS without having to resort to 0globals.conf.

instantlinux, 2023-01-25:
Feature is added, pull the image `instantlinux/samba-dc:4.16.8-r0` to pick up this capability.

bhechinger, 2023-02-16:
I think this maybe isn't working correctly? My dns forwarding isn't working and if I look at my smb.conf I see this:

```
# Generated by entrypoint.sh. Add customizations under /etc/samba/conf.d.
#  DO NOT EDIT THIS FILE.

[global]
        netbios name = DC1
        realm = 4AMLUNCH.NET
        server role = active directory domain controller
        workgroup = 4AMLUNCH

        add machine script = /usr/sbin/adduser -D -H -G users -s /bin/false %u
        allow dns updates = secure
        bind interfaces only = yes
        domain logons = yes
        domain master = no
        interfaces = internal
        log level = 1
        winbind refresh tickets = Yes
        winbind use default domain = yes
{% if DNS_FORWARDER %}
    dns forwarder = {{ DNS_FORWARDER }}
{%  endif %}
include = /etc/samba/conf.d/netlogon.conf
include = /etc/samba/conf.d/sysvol.conf
```

Shouldn't it have parsed that as part of the template and put the actual values in there?

bhechinger, 2023-02-16:
Oh, I found #106 never mind. :-D

closed 2025-09-25 by instantlinux