instantlinux/docker-tools #46: system crash on startup

jmls, 2020-09-04:
I'm trying to start the container with the default docker-compose and get these errors

```
app_1  | Set timezone
app_1  | INFO 2020-09-04 13:00:54,552 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2133: Looking up IPv4 addresses
app_1  | WARNING 2020-09-04 13:00:54,569 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2143: No IPv4 address will be assigned

```

followed by

```
app_1  | ERROR(runtime): uncaught exception - (3221225506, '{Access Denied} A process has requested access to an object but has not been granted those access rights.')
app_1  |   File "/usr/lib/python3.8/site-packages/samba/netcmd/__init__.py", line 186, in _run
app_1  |     return self.run(*args, **kwargs)
app_1  |   File "/usr/lib/python3.8/site-packages/samba/netcmd/domain.py", line 487, in run
app_1  |     result = provision(self.logger,
app_1  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 2353, in provision
app_1  |     provision_fill(samdb, secrets_ldb, logger, names, paths,
app_1  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1990, in provision_fill
app_1  |     setsysvolacl(samdb, paths.netlogon, paths.sysvol, paths.root_uid,
app_1  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1770, in setsysvolacl
app_1  |     _setntacl(sysvol)
app_1  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1764, in _setntacl
app_1  |     return setntacl(
app_1  |   File "/usr/lib/python3.8/site-packages/samba/ntacls.py", line 236, in setntacl
app_1  |     smbd.set_nt_acl(

```

and the container then continuously restarts

Am I missing some config options (I did change action to provision)

Roemer, 2021-02-19:
I am having the same issue. Not sure what is wrong. could you get it to run?

Roemer, 2021-02-19:
I fixed that issues by using `privileged: true` in the compose file and also set the environment variable `INTERFACES` to `lo ens192` but now it is throwing errors about dns updates (`ERROR(runtime): uncaught exception - (9711, 'WERR_DNS_ERROR_RECORD_ALREADY_EXISTS')`) and I cannot connect to the DC.

instantlinux, 2022-01-09:
Thanks for this report! I've updated the example `docker-compose.yml` and mentioned the kernel capability requirement in the README.

closed 2022-01-09 by instantlinux