instantlinux/docker-tools #48: Running mythtv container crashes on mythtv-setup step

robbadler, 2020-10-26:
Using `image: instantlinux/mythtv-backend:31.0-fixes.202007040047.3ef7db6` I am getting a crash on the `mythtv-setup` step, with the terminal log:
```
mythtv@testmyth31:~$ mythtv-setup --logpath $(pwd) --geometry=1280x720
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
mythtv@testmyth31:~$
```

My docker-compose is thus:
```
version: '3.3'

services:
  mythtv31:
    image: instantlinux/mythtv-backend:31.0-fixes.202007040047.3ef7db6
    hostname: testmyth31
    container_name: testmyth31
    depends_on: [mariadb]
    restart: always
    environment:
      DBNAME: mythconverg
      DBPASSWORD: mythtv
      DBSERVER: mariadb
      LOCALHOSTNAME: testmyth31
      TZ: 'Americas/Los_Angeles'
    volumes:
    - apache2:/var/log/apache2
    - data:/var/mythdata
    - home:/home/mythtv
    - ssh:/etc/ssh
    - /var/mythtv:/var/mythtv
    secrets: []
    #- mythtv-db-password
    #- mythtv-user-password
    mac_address: 02:42:c0:a8:01:03
    networks:
    - lan

  mariadb:
    image: mariadb
    container_name: mariadb
    restart: always
    volumes:
    - maria_datadir:/var/lib/mysql
    - /media/mirror/incoming/mythbackup:/mythbackup:ro
    environment:
      MYSQL_ROOT_PASSWORD: hjaebl
    networks:
    - lan
    - default

secrets:
  mythtv-db-password:
    file: ./secrets/mythtv-db-password
  mythtv-user-password:
    file: ./secrets/mythtv-user-password

networks:
  lan:
    #name: lan
    driver: macvlan
    driver_opts:
      parent: eno2
    ipam:
      config:
      - subnet: 192.168.1.2/29
volumes:
  apache2:
  data:
  home:
  ssh:
  maria_datadir:
    driver: local
    driver_opts:
      type: none
      device: /media/disk/docker_compose/mythtv-compose/mysql_data
      o: bind
```

instantlinux, 2021-05-09:
This message is expected and unrelated to your issue:
```
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```
I re-tested this with the current version of the container and it works the same as always, with the instructions from the README. After connecting with ssh, make sure the DISPLAY variable matches the one on your host, and that you have X11 (or the equivalent if using a Mac) running.
```
docker exec mythtvbackend_app_1 /usr/sbin/sshd
ssh -p 2022 -X mythtv@<host-ip>
mythtv@pvr01:~$ echo $DISPLAY
localhost:11.0
mythtv@pvr01:~$ mythtv-setup
```

instantlinux, 2021-11-27:
Closing, not able to reproduce.

closed 2021-11-27 by instantlinux