instantlinux/docker-tools #54: Unable to execute nut-upsd on a raspberry pi 4

Edo78, 2021-01-01:
I'm unable to use the image `instantlinux/nut-upsd`
```
$ docker-compose up nut
Creating nut ... done
Attaching to nut
nut               | Error relocating /usr/sbin/upsdrvctl: __select_time64: symbol not found
nut               | Error relocating /usr/sbin/upsdrvctl: __difftime64: symbol not found
nut               | Error relocating /usr/sbin/upsdrvctl: __gettimeofday_time64: symbol not found
nut               | Error relocating /usr/sbin/upsdrvctl: __stat_time64: symbol not found
nut exited with code 127
```
but if I use the old `instantlinux/nut-upsd-arm32` I'm able to have it running just fine
```
$ docker-compose up nut
Recreating nut ... done
Attaching to nut
nut               | Network UPS Tools - UPS driver controller 3.8.0-3727-geade014bef
nut               | Network UPS Tools - Generic HID driver 0.41 (3.8.0-3727-geade014bef)
nut               | Using subdriver: APC HID 0.96
nut               | USB communication driver 0.33
nut               | Ignoring invalid pid number 0
nut               | Network UPS Tools upsd 3.8.0-3727-geade014bef
nut               | /etc/nut/upsd.conf is world readable
nut               | listening on 0.0.0.0 port 3493
nut               | Connected to UPS [ups]: usbhid-ups-ups
nut               | /etc/nut/upsd.users is world readable
nut               | Network UPS Tools upsmon 3.8.0-3727-geade014bef
nut               |    0.000000 Ignoring invalid pid number 0
nut               |    0.000133 UPS: ups@localhost (master) (power value 1)
nut               |    0.000170 Using power down flag file /etc/killpower
nut               |    0.000250 debug level is '1'
nut               |    0.007284 Trying to connect to UPS [ups@localhost]
nut               |    0.008410 Logged into UPS ups@localhost
```
In my `docker-compose.yaml` I have
```
  nut:
    container_name: nut
    image: instantlinux/nut-upsd-arm32
#    image: instantlinux/nut-upsd
    volumes:
      - ./nut:/etc/nut
    ports:
      - 3493:3493
    secrets:
      - nut-upsd-password
    devices:
      - /dev/bus/usb:/dev/bus/usb
    privileged: true
```

instantlinux, 2021-01-03:
Thanks for the report! This is likely an issue with the upstream alpine package build, will have to reproduce and and see if I can come up with a workaround.

Edo78, 2021-01-03:
Let me know if I can provide any info to help you solve this issue

instantlinux, 2021-01-14:
Found a report in the upstream repo: https://gitlab.alpinelinux.org/alpine/aports/-/issues/12182. Followed the advice there and built an updated image. Pull the image and try again:
```
docker pull instantlinux/nut-upsd:2.7.4-r7
```

Edo78, 2021-01-15:
I can confirm that `instantlinux/nut-upsd:2.7.4-r7` is working on my `armv7` rpi4

closed 2021-01-17 by instantlinux