instantlinux/docker-tools #56: spamassassin error

fabriciomrtnz, 2021-01-28:
Im trying to use the spamassassin image and I get the following errors:

Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/Razor2/Client/Config.pm line 442.
Use of uninitialized value in concatenation (.) or string at /usr/share/perl5/Razor2/Client/Config.pm line 443.

Register successful.  Identity stored in /var/lib/spamassassin/.razor/identity-ruxMYI9zUs
Jan 28 18:05:44.617 [27] error: Can't locate Mail/SpamAssassin/CompiledRegexps/body_0.pm in @INC (you may need to install the Mail::SpamAssassin::CompiledRegexps::body_0 module) (@INC contains: /var/lib/spamassassin/compiled/5.028/3.004002 /var/lib/spamassassin/compiled/5.028/3.004002/auto lib /usr/share/perl5 /etc/perl /usr/local/lib/aarch64-linux-gnu/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/aarch64-linux-gnu/perl5/5.28 /usr/lib/aarch64-linux-gnu/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/aarch64-linux-gnu/perl-base) at (eval 1867) line 1.

not sure if its related to my setup or something wrong in the image.

instantlinux, 2021-02-26:
The first two warnings are expected (uninitialized value in Razor2) and cause no issue. I'm unfamiliar with the other error message you've reported. Is this a new spamassassin installation starting from scratch, or are you trying to replace an existing one with docker?

instantlinux, 2021-05-07:
Closing due to inactivity. See similar past reports in other distributions:
[ubuntu #1577925](https://bugs.launchpad.net/ubuntu/+source/spamassassin/+bug/1577925)
[untangle 05-20-2009](https://forums.untangle.com/spam-blocker-lite/9414-spamassassin-cant-locate-compiledregexps-body_0-pm.html)
[nabble 12-12-2015](https://forums.untangle.com/spam-blocker-lite/9414-spamassassin-cant-locate-compiledregexps-body_0-pm.html)
[debian #986034](https://groups.google.com/g/linux.debian.bugs.dist/c/ALdk5aAR-80)

Most likely, the command `sa-update && sa-compile` will eliminate the issue if it was caused by a bad daily rule-set that got downloaded.

closed 2021-05-07 by instantlinux