instantlinux/docker-tools #60: nut-upsd does not work with snmp-ups driver

boingoing, 2021-02-23:
The driver is here in the container but the plumbing seems wrong.

Trying to use docker-compose with this config:
```
version: '3.6'
services:
  nut-upsd:
    container_name: nut-upsd
    image: instantlinux/nut-upsd
    volumes:
      - ${USERDIR}/docker/nut-upsd/config:/etc/nut/local
    ports:
      - 3493:3493
    restart: unless-stopped
    environment:
      PORT: 192.168.1.8
      DRIVER: snmp-ups
    secrets:
      - nut-upsd-password
```

Output is:
```
Recreating nut-upsd ... done
Attaching to nut-upsd
nut-upsd            | ** This container may not work without setting for SERIAL **
nut-upsd            | Network UPS Tools - UPS driver controller 20200917-3625-gd7fa06b609
nut-upsd            | Network UPS Tools - Generic SNMP UPS driver 0.97 (20200917-3625-gd7fa06b609)
nut-upsd            |
nut-upsd            | Fatal error: 'serial' is not a valid variable name for this driver.
nut-upsd            |
nut-upsd            | Look in the man page or call this driver with -h for a list of
nut-upsd            | valid variable names and flags.
nut-upsd            | Network UPS Tools upsd 20200917-3625-gd7fa06b609
nut-upsd            | Ignoring invalid pid number 0
nut-upsd            | listening on 0.0.0.0 port 3493
nut-upsd            | Can't connect to UPS [ups] (snmp-ups-ups): No such file or directory
nut-upsd            | Network UPS Tools upsmon 20200917-3625-gd7fa06b609
nut-upsd            |    0.000000       Ignoring invalid pid number 0
nut-upsd            |    0.000439       Using power down flag file /etc/killpower
nut-upsd            |    0.000693       UPS: ups@localhost (master) (power value 1)
nut-upsd            |    0.000845       debug level is '1'
nut-upsd            |    0.009620       Trying to connect to UPS [ups@localhost]
nut-upsd            |    0.010905       Logged into UPS ups@localhost
nut-upsd            |    0.011116       Poll UPS [ups@localhost] failed - Driver not connected
nut-upsd            |    0.011129       Communications with UPS ups@localhost lost
nut-upsd            | sh: wall: not found
nut-upsd            |    5.011774       Poll UPS [ups@localhost] failed - Driver not connected
nut-upsd            |    5.011825       UPS ups@localhost is unavailable
nut-upsd            | sh: wall: not found
nut-upsd            |   10.012666       Poll UPS [ups@localhost] failed - Driver not connected
```

If you can point me to the place in the repo where these arguments are plugged-in to the call to startup NUT, I can make an attempt to fix it though I am not very familiar.

instantlinux, 2021-02-26:
In the README, this is the likely solution:
   > If you have a different model of UPS, contents of the files ups.conf, upsd.conf, upsmon.conf, and/or upsd.users can be overridden by mounting them to /etc/nut/local

Look in the [entrypoint.sh](https://github.com/instantlinux/docker-tools/blob/master/images/nut-upsd/entrypoint.sh) container startup script to see how those files are generated.  In particular, at line 13 of that script, the `ups.conf` file is defined with a serial number. If you make your own copy of that file (omitting the `serial` directive) you should be able to get this going by volume-mounting your local copy into the container as `/etc/nut/ups.conf`.

boingoing, 2021-02-27:
Thanks for pointing me in the right direction. Setting up ups.conf and upsmon.conf in the mounted volume worked well and the container is now doing what it's supposed to be doing. 👍

closed 2021-02-27 by boingoing

juniorUniv, 2022-04-25:
Hi @instantlinux,
same issue here.

Background:
- Synology NAS is running with NUT (disabled NUT server feature)
- port 3493 exposed by localhost
- snmp-ups driver is running
- upsc ups@localhost .... is working as expected (connected to the UPS status by the snmp-ups)
- following the instruction here - container volume mounted from container source :/etc/nut
- all the conf are now available in the mounted volume
- checked & edited both: ups.conf and upsmon.conf (mounted volume) with existing files from the NAS (NUT localhost)

Still the same output:

```
Network UPS Tools - UPS driver controller 3.15.0_alpha20210804-3402-gced1683082
Network UPS Tools - Generic SNMP UPS driver 0.97 (3.15.0_alpha20210804-3402-gced1683082)

Fatal error: 'serial' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
Network UPS Tools upsd 3.15.0_alpha20210804-3402-gced1683082
Ignoring invalid pid number 0
/etc/nut/upsd.conf is world readable
listening on 0.0.0.0 port 3493
Can't connect to UPS [ups] (snmp-ups-ups): No such file or directory
/etc/nut/upsd.users is world readable
Network UPS Tools upsmon 3.15.0_alpha20210804-3402-gced1683082
   0.000000	Ignoring invalid pid number 0
   0.000411	UPS: ups@localhost:3493 (master) (power value 1)
   0.000618	Using power down flag file /etc/killpower
   0.000959	debug level is '1'
   0.002807	Init SSL without certificate database
   0.008102	Trying to connect to UPS [ups@localhost:3493]
   0.009488	Login on UPS [ups@localhost:3493] failed - got [ERR ACCESS-DENIED]
   5.011128	Poll UPS [ups@localhost:3493] failed - Driver not connected
  10.012519	Poll UPS [ups@localhost:3493] failed - Driver not connected
Executing command: nocomm
sh: /usr/syno/bin/synoups: not found
exec_cmd(/usr/syno/bin/synoups nocomm) returned 127
  15.015521	Poll UPS [ups@localhost:3493] failed - Driver not connected
  20.017514	Poll UPS [ups@localhost:3493] failed - Driver not connected
  25.018528	Poll UPS [ups@localhost:3493] failed - Driver not connected
```

Checked upsd.user at the container mounted volume for the "got [ERR ACCESS-DENIED]" ... "psw and user are defined correctly"
From the NAS console - ups@localhost is accessible

My compose file:
```
version: "3.1"

services:
  app:
    image: ${REGISTRY_URI:-instantlinux}/nut-upsd:latest
    container_name: nut-server
    volumes:
      - _myvolume_:/etc/nut
    restart: always
    environment:
      API_USER: master
      GROUP: 100
      NAME: ups
      DESCRIPTION: DK Plus 3000
      DRIVER: snmp-ups
      SERVER: master
      PORT: _IP_address_of_my_UPS_
      SDORDER: -1
      TZ: _MyTZvalue_
    ports:
    - 3494:3493
    network_mode: bridge
    secrets:
    - nut-upsd-password
    privileged: true

secrets:
  nut-upsd-password:
    file: _path_
```

Thx for a hint