instantlinux/docker-tools #67: nut-upsd connection lost

poliant, 2021-04-16:
I'm using nut-upsd image on a RPI4 for a tecnoware Era Plus 1100 and evything works fine except in the following cases:
 1) in case I disconnect and reconnect the USB on RPI
 2) UPS for some strange reason loses serial connection for a while

I both case the port changes (e.g. from /dev/bus/usb/00**X**/00**Z** to /dev/bus/usb/00**Y**/00**W**).
In order to have always the same port I've created a rule so that the usb port is always mapped to /dev/myUPS and I've configured such port on my dockerfile:
```
  nut-upsd:
    container_name: nut-upsd
    image: instantlinux/nut-upsd
    volumes:
      - "/home/antonio/nut-upsd-pwd:/run/secrets/nut-upsd-password"
    environment:
    - "PORT=/dev/myUPS"
    - "NAME=tecnoware"
    - "DESCRIPTION=Technoware Era Plus 1100"
    - "DRIVER=blazer_usb"
    - "API_USER=master"
    network_mode: host
    privileged: true
    restart: always
```

Also with such configuration I lose the connection to UPS in both cases I reported above... of course the first case is not real, meaning I used for testing... of course I've checked that when the USB is reconnected and/or the UPS re-establish the connection, symbolik link to usb port is updated correctly.

Thanks for support

hellcry37, 2023-12-16:
Hello,
I am using your docker container that worked well til the latest update or so. I am guessing is the same as the author, I did not want to hijack his tread.

```
  nut-server:
    image: instantlinux/nut-upsd:latest
    container_name: nut-server
    restart: unless-stopped
    privileged: true
    environment:
      SERIAL: xxxxxxxxxxxx
      DRIVER: usbhid-ups
      NAME: CP1500EPFCLCD
      DESCRIPTION: Cyber Power CP1500 AVR UPS
    ports:
      - 3493:3493
    networks:
      - smarthome
    secrets:
      - nut-upsd-password
```

after a while:
```
33968.169374    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33973.169865    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33978.170490    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33983.171079    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33988.171785    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33993.172246    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
33998.172700    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34003.173292    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34008.173541    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34013.174100    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34018.174301    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34023.174886    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34028.175143    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34033.175663    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34038.175878    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34043.176376    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
34048.176594    Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
```

after more time:
```
34878.245858    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34883.249626    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34883.249644    UPS CP1500EPFCLCD@localhost is unavailable
sh: wall: not found
34888.250321    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34893.253477    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34898.253745    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34903.255366    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34908.255462    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34913.257664    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34918.257764    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34923.261468    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34928.261587    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34933.265438    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34938.265570    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
34943.269462    UPS [CP1500EPFCLCD@localhost]: connect failed: Server disconnected
34948.269659    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
Error while connecting to NUT server localhost, disconnect
[... 259 lines cut ...]
47433.917583    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
47564.089333    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
47569.089527    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
47699.253521    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
47704.253745    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
47834.421593    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
47834.421628    UPS CP1500EPFCLCD@localhost is unavailable
sh: wall: not found
47839.422360    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
47969.589244    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
47974.589438    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
48104.757549    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
48109.757741    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
48239.925547    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
48239.925601    UPS CP1500EPFCLCD@localhost is unavailable
sh: wall: not found
48244.926421    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
48375.093593    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
48380.093784    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
48510.261493    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
48515.261625    [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
48645.429351    UPS [CP1500EPFCLCD@localhost]: connect failed: Connection failure: Address not available
48645.429395    UPS CP1500EPFCLCD@localhost is unavailable
sh: wall: not found
```

instantlinux, 2023-12-16:
What is the output of this command (part of Dockerfile's `HEALTHCHECK`) directive, when this happens with your UPS model, from a `docker exec -it <id> sh` shell?
```
upsc $NAME@localhost:3493 2>&1
```
An important assumption is that the `upsc` command output will include the word `stale`, and cause the container to restart: but evidently the output is vendor-specific, so the pattern to be matched needs updating. Please help out by providing the output for the Cyber Power.

hellcry37, 2023-12-17:
hello,

Here is the required information:

/ # echo $NAME
CP1500EPFCLCD
/ # upsc $CP1500EPFCLCD@localhost:3493 2>&1
upscli_splitname: got empty upsname string
Error: invalid UPS definition.
Required format: upsname[@hostname[:port]]

hellcry37, 2023-12-17:
I want to mention that nut-upsd:2.8.0-r4 works ok and has the following output:

```
/ # echo $NAME
CP1500EPFCLCD
/ # upsc $CP1500EPFCLCD@localhost:3493 2>&1
Init SSL without certificate database
Error: Unknown UPS
```

instantlinux, 2023-12-17:
The command output I want to see is
```
upsc $NAME@localhost:3493 2>&1
```
which expands the variable $NAME, not expanding a non-existent variable $CP1500EPFCLCD, below is incorrect:
```
upsc $CP1500EPFCLCD@localhost:3493 2>&1
```
Someone reported a bug with there being two `$$` characters in the healthcheck: this was corrected in the `2.8.1-r0` tag; I've yet to fully confirm whether the fix (PR #140) is proper. To confirm, you can do `docker history instantlinux/nut-upsd:<tag>` for that tag or others that are present on your host.

Does the symptom still appear when you comment out the _NAME: CP1500EPFCLCD_ environment variable in docker-compose.yml?

hellcry37, 2023-12-18:
old 2.8.0-r4:

```txt
/ # upsc $NAME@localhost:3493 2>&1
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 3150
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: CRMKT2000280
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.serial: CRMKT2000280
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 232.0
input.voltage.nominal: 230
output.voltage: 232.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 13
ups.mfr: CPS
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CRMKT2000280
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

new latest 2.8.0-r0:
```txt
/ # upsc $NAME@localhost:3493 2>&1
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 3120
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: CRMKT2000280
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.serial: CRMKT2000280
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 230.0
input.voltage.nominal: 230
output.voltage: 230.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 13
ups.mfr: CPS
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: CRMKT2000280
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```

hellcry37, 2023-12-18:
I'll test this now?

```txt
Does the symptom still appear when you comment out the NAME: CP1500EPFCLCD environment variable in docker-compose.yml?
```

hellcry37, 2023-12-18:
after not even a few minutes:

logs: docker logs nut-server
```txt
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
Using subdriver: CyberPower HID 0.8
USB communication driver (libusb 1.0) 0.46
WARNING: Needed to fix group access to filesystem socket of this driver, but failed; run the driver with more debugging to see how exactly.
Consumers of the socket, such as upsd data server, can fail to interact with the driver and represent the device: /var/run/nut/usbhid-ups-ups
Ignoring invalid pid number 0
Network UPS Tools upsd 2.8.1
listening on 0.0.0.0 port 3493
Connected to UPS [ups]: usbhid-ups-ups
Found 1 UPS defined in ups.conf
Network UPS Tools upsmon 2.8.1
   0.000000     Ignoring invalid pid number 0
   0.000028     [D1] Just failed to send signal, no daemon was running
   0.000189     Using power down flag file /etc/killpower
   0.000365     UPS: ups@localhost (primary) (power value 1)
   0.000396     [D1] debug level is '1'
   0.000621     [D1] Saving PID 25 into /run/upsmon.pid
   0.000789     [D1] Succeeded to become_user(nut): now UID=100 GID=101
Init SSL without certificate database
   0.001894     upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.001904     [D1] Trying to connect to UPS [ups@localhost]
   0.002217     [D1] Logged into UPS ups@localhost
 565.061320     Poll UPS [ups@localhost] failed - Data stale
 565.061394     Communications with UPS ups@localhost lost
sh: wall: not found
 570.062669     Poll UPS [ups@localhost] failed - Data stale
 575.063161     Poll UPS [ups@localhost] failed - Data stale
 580.063387     Poll UPS [ups@localhost] failed - Data stale
 585.063580     Poll UPS [ups@localhost] failed - Data stale
 590.064142     Poll UPS [ups@localhost] failed - Data stale
 595.064605     Poll UPS [ups@localhost] failed - Data stale
 600.065096     Poll UPS [ups@localhost] failed - Data stale
 605.065808     Poll UPS [ups@localhost] failed - Data stale
 610.066411     Poll UPS [ups@localhost] failed - Data stale
 615.066849     Poll UPS [ups@localhost] failed - Data stale
 620.067402     Poll UPS [ups@localhost] failed - Data stale
 625.067720     Poll UPS [ups@localhost] failed - Data stale
 630.068213     Poll UPS [ups@localhost] failed - Data stale
 635.068767     Poll UPS [ups@localhost] failed - Data stale
 640.069205     Poll UPS [ups@localhost] failed - Data stale
```

instantlinux, 2023-12-19:
@hellcry37 @poliant @KC-inDomus please invoke `docker pull instantlinux/nut-upsd:latest` (or tag `2.8.1-r0` to update if you use this tag) and confirm the two changes I've made. The first change is to ensure that the container takes itself down for restart if data goes stale, and the second is to add the `MAXAGE` parameter. If you're using Cyberpower model, add this to your docker-compose file in the nut-server -> environment section:
```
    environment:
      ...
      MAXAGE: 25
```

hellcry37, 2023-12-19:
Does this fix the issue or just restarts when "Data stale" is detected? If the issue is not fixed that means container will restart in a loop at some point (that will not help)

what is that MAXAGE: 25 do?

instantlinux, 2023-12-19:
If the issue is caused by the UPS unit taking longer than 15 seconds, the default value of MAXAGE, to report its data: it will be fixed. Please change that parameter and see if the container stays running without restarts. Googling for this problem is inconclusive.

hellcry37, 2023-12-19:
will change it now and come back

hellcry37, 2023-12-19:
Til now it seem to work ok.

However we have a warning I dont know if is of any importance:
```txt
WARNING: Needed to fix group access to filesystem socket of this driver, but failed; run the driver with more debugging to see how exactly.
```

I'll come back tomorrow to tell you if this worked as it should. Thank you for all your hard work and your patience!

hellcry37, 2023-12-19:
Ahhh too soon it just failed again container restarted:
```txt
74afb66911c9   instantlinux/nut-upsd:latest             "/bin/sh -c /usr/loc…"   29 minutes ago   Up 59 seconds (healthy)   0.0.0.0:3493->3493/tcp, :::3493->3493/tcp      nut-server
```

```txt
docker logs nut-server
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
Using subdriver: CyberPower HID 0.8
USB communication driver (libusb 1.0) 0.46
WARNING: Needed to fix group access to filesystem socket of this driver, but failed; run the driver with more debugging to see how exactly.
Consumers of the socket, such as upsd data server, can fail to interact with the driver and represent the device: /var/run/nut/usbhid-ups-CP1500EPFCLCD
Network UPS Tools upsd 2.8.1
Ignoring invalid pid number 0
listening on 0.0.0.0 port 3493
Connected to UPS [CP1500EPFCLCD]: usbhid-ups-CP1500EPFCLCD
Found 1 UPS defined in ups.conf
Network UPS Tools upsmon 2.8.1
   0.000000     Ignoring invalid pid number 0
   0.000028     [D1] Just failed to send signal, no daemon was running
   0.000403     Using power down flag file /etc/killpower
   0.000758     UPS: CP1500EPFCLCD@localhost (primary) (power value 1)
   0.000792     [D1] debug level is '1'
   0.001066     [D1] Saving PID 26 into /run/upsmon.pid
   0.001247     [D1] Succeeded to become_user(nut): now UID=100 GID=101
Init SSL without certificate database
   0.002293     upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.002301     [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
   0.002624     [D1] Logged into UPS CP1500EPFCLCD@localhost
1695.185795     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
1695.185828     Communications with UPS CP1500EPFCLCD@localhost lost
sh: wall: not found
1700.186960     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
1705.187683     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
1710.188270     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
1715.188906     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
1718.291046     Signal 15: exiting
1718.292958     upsmon parent: read
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
Using subdriver: CyberPower HID 0.8
USB communication driver (libusb 1.0) 0.46
WARNING: Needed to fix group access to filesystem socket of this driver, but failed; run the driver with more debugging to see how exactly.
Consumers of the socket, such as upsd data server, can fail to interact with the driver and represent the device: /var/run/nut/usbhid-ups-CP1500EPFCLCD
Ignoring invalid pid number 0
Network UPS Tools upsd 2.8.1
listening on 0.0.0.0 port 3493
Connected to UPS [CP1500EPFCLCD]: usbhid-ups-CP1500EPFCLCD
Found 1 UPS defined in ups.conf
Network UPS Tools upsmon 2.8.1
   0.000000     Ignoring invalid pid number 0
   0.000068     [D1] Just failed to send signal, no daemon was running
   0.000312     Using power down flag file /etc/killpower
   0.000451     UPS: CP1500EPFCLCD@localhost (primary) (power value 1)
   0.000494     [D1] debug level is '1'
   0.000929     [D1] Saving PID 19 into /run/upsmon.pid
   0.001216     [D1] Succeeded to become_user(nut): now UID=100 GID=101
Init SSL without certificate database
   0.002909     upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.002927     [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
   0.003618     [D1] Logged into UPS CP1500EPFCLCD@localhost
 125.016864     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 125.016908     Communications with UPS CP1500EPFCLCD@localhost lost
sh: wall: not found
 130.017894     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 135.018449     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 140.019069     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 145.019420     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 150.019670     Poll UPS [CP1500EPFCLCD@localhost] failed - Data stale
 150.633707     Signal 15: exiting
 150.635504     upsmon parent: read
Network UPS Tools - UPS driver controller 2.8.1
Network UPS Tools - Generic HID driver 0.52 (2.8.1)
USB communication driver (libusb 1.0) 0.46
Using subdriver: CyberPower HID 0.8
WARNING: Needed to fix group access to filesystem socket of this driver, but failed; run the driver with more debugging to see how exactly.
Consumers of the socket, such as upsd data server, can fail to interact with the driver and represent the device: /var/run/nut/usbhid-ups-CP1500EPFCLCD
Ignoring invalid pid number 0
Network UPS Tools upsd 2.8.1
listening on 0.0.0.0 port 3493
Connected to UPS [CP1500EPFCLCD]: usbhid-ups-CP1500EPFCLCD
Found 1 UPS defined in ups.conf
   0.000000     Ignoring invalid pid number 0
Network UPS Tools upsmon 2.8.1
   0.000062     [D1] Just failed to send signal, no daemon was running
   0.000482     Using power down flag file /etc/killpower
   0.000698     UPS: CP1500EPFCLCD@localhost (primary) (power value 1)
   0.000770     [D1] debug level is '1'
   0.001433     [D1] Saving PID 19 into /run/upsmon.pid
   0.001898     [D1] Succeeded to become_user(nut): now UID=100 GID=101
Init SSL without certificate database
   0.004543     upsnotify: failed to notify about state 2: no notification tech defined, will not spam more about it
   0.004565     [D1] Trying to connect to UPS [CP1500EPFCLCD@localhost]
   0.005634     [D1] Logged into UPS CP1500EPFCLCD@localhost
```

instantlinux, 2023-12-20:
In the container under directory /etc/nut, you'll find the config files. Copy those off into a directory on your local host (`docker cp /etc/nut/* .` or something like that) and then add a mount definition for that path:
```
  services:
    app:
      ...
      volumes:
      - ~/my-test-dir:/etc/nut/local
```
Check that the value of MAXAGE in `upsd.conf` is set to your desired value (if the entrypoint script didn't work, it'll still be defaulted to 15 instead of overridden to 25). Once you're running with files in the local override path, then you can experiment with values such as `POLLFREQ` and `DEADTIME` which I don't try to manage with the current input variables (which work fine for APC and some other models, but now we're trying to get CyberPower working).

How often is the restart happening, after the first hour or so of running? If changes to some of those other variables has any effect, can you get it to stay stable for longer than several hours at a time? I'll be happy to add support for more variables once this is confirmed.

hellcry37, 2023-12-20:
I did not see a point in modify so much the docker to make a volume and add files there, I did all this in the container to get data for you.

docker exec -it nut-server sh
/ # cd /etc/nut/
/etc/nut # cat upsd.conf

```txt
# =======================================================================
# MAXAGE <seconds>
MAXAGE 25
```
So from here we conclude the MAXAGE var works fine.

Container restarts one per 1h / 3h is not a well defined time tbh. In HA it dropped on:
16:41
19:39
20:32
23:40
05:09

As you can see not a well defined pattern :)

hellcry37, 2023-12-20:
switched to local config files mode! What should I test next?

instantlinux, 2023-12-21:
I can't think of what specific changes might address the problem. Searching online, I found a few discussions about the CyberPower stale-data problem posted over the past several years:

* [StackExchange #66611](https://raspberrypi.stackexchange.com/questions/66611/nut-cyberpower-data-stale)
* [reddit/r/homelab](https://www.reddit.com/r/homelab/comments/112lfon/cant_get_my_cyberpower_ups_to_stay_connected_to/)
* [hassio-addons \#91](https://github.com/hassio-addons/addon-nut/issues/91)

Those, along with comments found in the distributed config files under /etc/nut, should help point you to the solutions others have found. Once you can get it stable 24hrs, post the resulting values here. Thanks!

hellcry37, 2023-12-22:
Whatever I do still fails, I'll just use the old one and move on with my life

instantlinux, 2023-12-24:
OK fair enough. Hopefully the restart logic is now stable enough to use, and that others will post their successful parameter values for this model or others that don't behave the same way as APC.

tabertantrum, 2024-02-07:
Hello @instantlinux, I've been facing the stale-data issue with two CyberPower UPS. The `MAXAGE` parameter does not appear to be working for me when creating the container. The `upsd.conf` file only includes: `LISTEN 0.0.0.0` after the commented sections from NUT.

I am mounting a custom `ups.conf` and `upsmon.conf` file to configure both UPS devices in one container, which has been working fine aside from the stale-data issue.

By creating and mounting a `upsd.conf` with this:
```
MAXAGE 60
LISTEN 0.0.0.0
```
Both of the UPS devices have been running for 36 hours continuously without issue. I may try to lower the `MAXAGE` value defined in the `upsd.conf` file, but I'm not familiar enough with shell to know why the if block in the entrypoint.sh for `MAXAGE` appears to be ignored. 

Here's the relevant details of my compose file in case this is helpful.
```
nut-upsd:
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    volumes:
      - /opt/nut:/etc/nut/local
    devices:
      - /dev/bus/usb
    environment:
      API_PASSWORD: "XXXXXXXXX"
      MAXAGE: 60
    restart: unless-stopped
```

I'd be happy to test this out further with you to confirm the `MAXAGE` variable is resolving the issue.

jkaberg, 2025-02-03:
> Hello [@instantlinux](https://github.com/instantlinux), I've been facing the stale-data issue with two CyberPower UPS. The `MAXAGE` parameter does not appear to be working for me when creating the [...]

I have the same manufacturer and the UPS PowerWalker VI 1200 SHL, the only thing that reliably works for me is to soft-reset the port using `usbreset`. I proposed to include this tool in https://github.com/instantlinux/docker-tools/issues/183. [Someone else](https://github.com/netinvent/usb_resetter) also faced similar issue with similar solution, so it seems CyberPower ships shitty HW/FW.

instantlinux, 2025-09-22:
Closing due to inactivity.

closed 2025-09-22 by instantlinux

instantlinux, 2025-09-22:
If you find a workaround that will help others @jkaberg , please raise a new issue.