instantlinux/docker-tools #68: How I configure 2 KeepAlived_HAProxy on Fault tolerance o Load balance

ManjiChiba, 2021-06-16:
Hi Rich:

I newbie on Docker and Docker-compose and don't know too much.

I try to do a 2 KeepAlived_HAProxy on my Raspbian 3B, but... I don't understand well how work your container.

If I put 
network_mode: host 
on my first haproxy_keep1.yml it appariency no fault, but... I don't understand about unicast on keepalived1.conf and do this error on log on Portainer/container

2021-06-16T19:52:41.838730+00:00 RN Jan  0 00:00:00 Keepalived[19]: Starting Keepalived v2.2.0 (01/10,2021), git commit v3.13.0_rc3-144-gef96292523
2021-06-16T19:52:41.838782+00:00 RN Keepalived[19]: Running on Linux 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 (built for Linux 5.7.8)
2021-06-16T19:52:41.838835+00:00 RN Keepalived[19]: Command line: 'keepalived' '-i' 'main'
2021-06-16T19:52:41.839225+00:00 RN Keepalived[19]: Configuration file /etc/keepalived/keepalived.conf
2021-06-16T19:52:41.843370+00:00 RN Keepalived[21]: NOTICE: setting config option max_auto_priority should result in better keepalived performance
2021-06-16T19:52:41.844559+00:00 RN Keepalived[21]: Starting VRRP child process, pid=23
2021-06-16T19:52:41.847144+00:00 RN Keepalived_vrrp[23]: bind unicast_src 172.22.1.33 failed 99 - Address not available
2021-06-16T19:52:41.847226+00:00 RN Keepalived_vrrp[23]: (haproxy_keep1): entering FAULT state (src address not configured)
2021-06-16T19:52:41.847253+00:00 RN Keepalived_vrrp[23]: (haproxy_keep1) Entering FAULT STATE
2021-06-16T19:52:42.834053+00:00 RN - 1684500332 19 : [origin software="rsyslogd" swVersion="8.2012.0" x-pid="16" x-info="https://www.rsyslog.com"] start

but, if I assing a ip to my container on haproxy_keep1.yml with 

    #networks:
      #contenedores:
        #ipv4_address: 172.22.1.33

The interface br-f1f707a85295
is blocked and the logs show me this another error

2021-06-16T20:23:03.527836+00:00 e54580a9db8a Jan  0 00:00:00 Keepalived[17]: Starting Keepalived v2.2.0 (01/10,2021), git commit v3.13.0_rc3-144-gef96292523
2021-06-16T20:23:03.527887+00:00 e54580a9db8a Keepalived[17]: Running on Linux 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 (built for Linux 5.7.8)
2021-06-16T20:23:03.527940+00:00 e54580a9db8a Keepalived[17]: Command line: 'keepalived' '-i' 'main'
2021-06-16T20:23:03.528342+00:00 e54580a9db8a Keepalived[17]: Configuration file /etc/keepalived/keepalived.conf
2021-06-16T20:23:03.530371+00:00 e54580a9db8a Keepalived[19]: NOTICE: setting config option max_auto_priority should result in better keepalived performance
2021-06-16T20:23:03.531636+00:00 e54580a9db8a Keepalived[19]: Starting VRRP child process, pid=21
2021-06-16T20:23:03.533473+00:00 e54580a9db8a - 2127257888 50 Keepalived_vrrp[21]: (/etc/keepalived/keepalived.conf: Line 22) WARNING - interface br-f1f707a85295 for vrrp_instance haproxy_keep1 doesn't exist
2021-06-16T20:23:03.533914+00:00 e54580a9db8a Keepalived_vrrp[21]: Non-existent interface specified in configuration
2021-06-16T20:23:03.534263+00:00 e54580a9db8a Keepalived_vrrp[21]: Stopped
2021-06-16T20:23:03.535322+00:00 e54580a9db8a Keepalived[19]: pid 21 exited with permanent error CONFIG. Terminating
2021-06-16T20:23:03.535900+00:00 e54580a9db8a Keepalived[19]: Stopped Keepalived v2.2.0 (01/10,2021), git commit v3.13.0_rc3-144-gef96292523
2021-06-16T20:23:04.522230+00:00 e54580a9db8a - 1684500332 19 : [origin software="rsyslogd" swVersion="8.2012.0" x-pid="14" x-info="https://www.rsyslog.com"] start

It is than if I deffined a ip for the container... the interface is blocked.

Another question...
Must be the unicast Ip addres the same than the container or another diferrent?
Where I must defined it?

arunvc, 2022-01-05:
Does it guarantee HA if haproxy container is down and dockerd is up ( examples/keepalived.conf)

instantlinux, 2023-10-25:
Closing due to inactivity.