instantlinux/docker-tools #69: Samba-AD-DC | No NTP/Windows Time provider

carev001, 2021-06-23:
I noticed the image currently does not provide NTP/Windows Time to domain-joined hosts. Any chance to incorporate that into the image?

thorus90, 2022-09-17:
I did run into the exact same issue, and was not able to fix it for myself, because you have two ways to provide NTP support for samba as DC:

- ntp server
- chrony

The ntp server is not part of alpine (only openntpd which does not work).
chrony is not compiled with the feature in alpine.

Any insight in how to incooperate this? Because right now all my windows machine don't have ntp syncing, because it requires signed NTP communication.

thorus90, 2022-09-29:
I got it to run, my initial problem was with the one windows client I tested it, all the other clients work just fine for about a week now. I tried to create a merge request. As it is my first merge request on github ever I don't know if I did everything correctly, feel free to correct it if need be.