instantlinux/docker-tools #70: ERROR(ldb): uncaught exception - Element clearTextPassword has empty attribute in ldb message (CN=Administrator,CN=Users,DC=ad,DC=***,DC=nl)!

Shadow7714, 2021-08-11:
Starting a fresh docker container keeps erroring out with:

> ...
> INFO 2021-08-11 23:56:45,149 pid:20 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1567: Setting up sam.ldb users and groups
> 
> 
> ERROR(ldb): uncaught exception - Element clearTextPassword has empty attribute in ldb message (CN=Administrator,CN=Users,DC=ad,DC=vlnet,DC=nl)!
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/netcmd/__init__.py", line 186, in _run
> 
> 
>     return self.run(*args, **kwargs)
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/netcmd/domain.py", line 491, in run
> 
> 
>     result = provision(self.logger,
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/provision/__init__.py", line 2324, in provision
> 
> 
>     provision_fill(samdb, secrets_ldb, logger, names, paths,
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/provision/__init__.py", line 1935, in provision_fill
> 
> 
>     samdb = fill_samdb(samdb, lp, names, logger=logger,
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/provision/__init__.py", line 1568, in fill_samdb
> 
> 
>     setup_add_ldif(samdb, setup_path("provision_users.ldif"), {
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/provision/common.py", line 55, in setup_add_ldif
> 
> 
>     ldb.add_ldif(data, controls)
> 
> 
>   File "/usr/lib/python3.9/site-packages/samba/__init__.py", line 229, in add_ldif
> 
> 
>     self.add(msg, controls)

I've tried deploying this on a fresh installed Ubuntu 20. VM, or directly on a Synology DS718+ NAS (in a macvlan network). In all cases I get the above error.

Run commands on Ubuntu VM:
`docker run --privileged --restart unless-stopped --network host --dns-search ad.vlnet.nl --dns 192.168.1.194 --hostname DC01 --name DC01 -e DOMAIN_ACTION=provision -e TZ='Europe/Amsterdam' -e NETBIOS_NAME=DC01 -e REALM=ad.vlnet.nl -v /mnt/nas3docker/DC01/data:/var/lib/samba -v /mnt/nas3docker/DC01/config/samba:/etc/samba -v /volume1/docker/DC01/secrets:/run/secrets -d instantlinux/samba-dc`

Run command on Synology NAS:
`docker run --privileged --network physical_network_noproxy --ip 192.168.1.195 --dns 192.168.1.194 --dns-search ad.vlnet.nl --dns-search ad.vlnet.nl --add-host DC01.ad.vlnet.nl:192.168.1.195 --hostname DC01 --name DC01 -e DOMAIN_ACTION=provision -e TZ='Europe/Amsterdam' -e NETBIOS_NAME=DC01 -e REALM=ad.vlnet.nl -e DOMAIN_LOGONS=no -v /volume1/docker/DC01/data:/var/lib/samba -v /volume1/docker/DC01/config:/etc/samba -v /volume1/docker/DC01/secrets:/run/secrets -d instantlinux/samba-dc`

instantlinux, 2022-01-09:
Attempted to reproduce (sorry for delay):
```
mkdir -p /var/tmp/issue70/data
mkdir -p /var/tmp/issue70/config
mkdir -p /var/tmp/issue70/secrets
echo -n '1foobar@' > /var/tmp/issue70/secrets/samba-admin-password
docker run --privileged --restart unless-stopped --network host \
  --dns-search ad.issue70.nl --dns 192.168.2.43 --hostname DC01 \
  --name DC01 -e DOMAIN_ACTION=provision -e TZ='Europe/Amsterdam' \
  -e NETBIOS_NAME=DC01 -e REALM=ad.issue70.nl \
  -v /var/tmp/issue70/data:/var/lib/samba \
  -v /var/tmp/issue70/config:/etc/samba \
  -v /var/tmp/issue70/secrets:/run/secrets \
  -d instantlinux/samba-dc:4.14.8-r0
```
Container started up OK, no errors:
```
INFO 2022-01-09 02:19:26,206 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2105: Looking up IPv4 addresses
WARNING 2022-01-09 02:19:26,646 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2115: No IPv4 address will be assigned
INFO 2022-01-09 02:19:26,646 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2122: Looking up IPv6 addresses
WARNING 2022-01-09 02:19:26,648 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2129: No IPv6 address will be assigned
INFO 2022-01-09 02:19:27,426 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2273: Setting up share.ldb
INFO 2022-01-09 02:19:27,454 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2277: Setting up secrets.ldb
INFO 2022-01-09 02:19:27,476 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2282: Setting up the registry
INFO 2022-01-09 02:19:27,553 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2285: Setting up the privileges database
INFO 2022-01-09 02:19:27,592 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2288: Setting up idmap db
INFO 2022-01-09 02:19:27,620 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #2295: Setting up SAM db
INFO 2022-01-09 02:19:27,628 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #880: Setting up sam.ldb partitions and settings
INFO 2022-01-09 02:19:27,630 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #892: Setting up sam.ldb rootDSE
INFO 2022-01-09 02:19:27,635 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1305: Pre-loading the Samba 4 and AD schema
Unable to determine the DomainSID, can not enforce uniqueness constraint on local domainSIDs

INFO 2022-01-09 02:19:27,683 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1383: Adding DomainDN: DC=ad,DC=issue70,DC=nl
INFO 2022-01-09 02:19:27,710 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1415: Adding configuration container
INFO 2022-01-09 02:19:27,732 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1430: Setting up sam.ldb schema
INFO 2022-01-09 02:19:34,198 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1448: Setting up sam.ldb configuration data
INFO 2022-01-09 02:19:34,532 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1489: Setting up display specifiers
INFO 2022-01-09 02:19:39,248 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1497: Modifying display specifiers and extended rights
INFO 2022-01-09 02:19:39,319 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1504: Adding users container
INFO 2022-01-09 02:19:39,322 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1510: Modifying users container
INFO 2022-01-09 02:19:39,324 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1513: Adding computers container
INFO 2022-01-09 02:19:39,327 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1519: Modifying computers container
INFO 2022-01-09 02:19:39,329 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1523: Setting up sam.ldb data
INFO 2022-01-09 02:19:39,645 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1553: Setting up well known security principals
INFO 2022-01-09 02:19:39,732 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1567: Setting up sam.ldb users and groups
INFO 2022-01-09 02:19:39,990 pid:16 /usr/lib/python3.9/site-packages/samba/provision/__init__.py #1575: Setting up self join
Repacking database from v1 to v2 format (first record CN=rpc-Profile,CN=Schema,CN=Configuration,DC=ad,DC=issue70,DC=nl)
Repack: re-packed 10000 records so far
Repacking database from v1 to v2 format (first record CN=rpcContainer-Display,CN=409,CN=DisplaySpecifiers,CN=Configuration,DC=ad,DC=issue70,DC=nl)
Repacking database from v1 to v2 format (first record CN=Guests,CN=Builtin,DC=ad,DC=issue70,DC=nl)
INFO 2022-01-09 02:19:42,528 pid:16 /usr/lib/python3.9/site-packages/samba/provision/sambadns.py #1143: Adding DNS accounts
INFO 2022-01-09 02:19:42,557 pid:16 /usr/lib/python3.9/site-packages/samba/provision/sambadns.py #1177: Creating CN=MicrosoftDNS,CN=System,DC=ad,DC=issue70,DC=nl
INFO 2022-01-09 02:19:42,600 pid:16 /usr/lib/python3.9/site-packages/samba/provision/sambadns.py #1190: Creating DomainDnsZones and ForestDnsZones partitions
INFO 2022-01-09 02:19:42,696 pid:16 /usr/lib/python3.9/site-packages/samba/provision/sambadns.py #1195: Populating DomainDnsZones and ForestDnsZones partitions
[... cut ...]
/usr/sbin/winbindd: daemon_ready: daemon 'winbindd' finished starting up and ready to serve connections
TLS self-signed keys generated OK
Doing a full scan on DC=ForestDnsZones,DC=ad,DC=issue70,DC=nl and looking for deleted objects
Doing a full scan on DC=DomainDnsZones,DC=ad,DC=issue70,DC=nl and looking for deleted objects
Doing a full scan on CN=Configuration,DC=ad,DC=issue70,DC=nl and looking for deleted objects
Doing a full scan on DC=ad,DC=issue70,DC=nl and looking for deleted objects
```
Perhaps your /volume1/docker/DC01/secrets/samba-admin-password file wasn't properly readable?

Steffenkt, 2022-05-22:
Since the problem is still marked as open, I might be able to help here.
In the data "/run/secrets/samba-admin-password" must be of course also the "samba-admin-password" included. If the file is empty, I get the same error message.

I think the problem can be closed. Maybe it just needs to be included in the documentation - Maybe I just overlooked it myself

instantlinux, 2022-05-22:
Thanks @Steffenkt, I added clarification to the [README](https://hub.docker.com/repository/docker/instantlinux/samba-dc).

closed 2022-08-27 by instantlinux