instantlinux/docker-tools #74: nut-upsd: 'No such file or directory' error when not using docker secrets

watson81, 2022-01-30:
In the NUT docker image, if a user chooses not to use Docker secrets, then the following error is emitted on startup:

> cat: can't open '/run/secrets/nut-upsd-password': No such file or directory

This error persists even when setting the `SECRET` environment variable to other values, though the file in the error message changes with the secret.

A user may choose to forgo using Docker secrets, or at least this Docker secret, if using custom configuration files.

instantlinux, 2022-04-27:
Implemented this conditional:
```
if [ -d /run/secrets ] && [ -s /run/secrets/$SECRET ]; then
  API_PASSWORD=$(cat /run/secrets/$SECRET)
fi
```
The API_PASSWORD could be set as env variable if not using a secret, or provided in a mounted config file.

bsmith76s, 2022-05-11:
> Implemented this conditional:
> 
> ```
> if [ -d /run/secrets ] && [ -s /run/secrets/$SECRET ]; then
>   API_PASSWORD=$(cat /run/secrets/$SECRET)
> fi
> ```
> 
> The API_PASSWORD could be set as env variable if not using a secret, or provided in a mounted config file.

How do I set as a variable I do not see that in documentation

instantlinux, 2022-05-13:
Doc is fixed now, thanks. See [README](https://hub.docker.com/r/instantlinux/nut-upsd).

closed 2022-05-13 by instantlinux