instantlinux/docker-tools #77: nut-upsd:  No matching HID UPS found

Impatience71, 2022-03-06:
Hi all,
last days I was successfull to configure upsd with the help of David Zomaya as stand alone version (not docker) - the trick was to not configure product and vendor ID.  The ups is a Eaton Ellipse Pro 1200.
I tried to hand over just a few parameters via the docker-compose, but the Driver failed to start. 
`nut:
    image: instantlinux/nut-upsd:latest
    environment:
      SERIAL:"G361M36084" 
      DRIVER:usbhid-ups 
      PORT:auto
    ports:
    - ${PORT_UPSD_1:-3493}:3493
    secrets:
    - nut-upsd-password
    privileged: true
`
The driver log 
`
No matching HID UPS found,
USB communication driver 0.33,
Driver failed to start (exit status=1),
Network UPS Tools - UPS driver controller 3.15.0_alpha20210804-3402-gced1683082,
Network UPS Tools - Generic HID driver 0.41 (3.15.0_alpha20210804-3402-gced1683082),
USB communication driver 0.33,
No matching HID UPS found,
Driver failed to start (exit status=1),
Network UPS Tools - UPS driver controller 3.15.0_alpha20210804-3402-gced1683082,
Network UPS Tools - Generic HID driver 0.41 (3.15.0_alpha20210804-3402-gced1683082),
USB communication driver 0.33,
No matching HID UPS found,
Driver failed to start (exit status=1)`

Any ideas what I made wrong ? 

Thanks Peter

instantlinux, 2022-05-05:
I can't offer help on a specific manufacturer's product; this one should work, though, with some details described [here](https://networkupstools.org/ddl/Eaton/Ellipse_PRO_650.html).

instantlinux, 2022-05-05:
By the wqy, use a tool like `yamllint` to make sure the yaml syntax is correct for docker-compose. With it I corrected some whitespace issues:
```
--- 
nut: 
  environment: 
    DRIVER: usbhid-ups
    PORT: auto
    SERIAL: G361M36084
  image: "instantlinux/nut-upsd:latest"
  ports: 
    - "${PORT_UPSD_1:-3493}:3493"
  privileged: true
  secrets: 
    - nut-upsd-password
```

instantlinux, 2022-05-05:
By the way, use a tool like `yamllint` to make sure the yaml syntax is correct for docker-compose. With it I corrected some whitespace issues:
```
--- 
nut: 
  environment: 
    DRIVER: usbhid-ups
    PORT: auto
    SERIAL: G361M36084
  image: "instantlinux/nut-upsd:latest"
  ports: 
    - "${PORT_UPSD_1:-3493}:3493"
  privileged: true
  secrets: 
    - nut-upsd-password
```

instantlinux, 2022-05-13:
Closing due to inactivity. Post here if you still cannot get your config working.

closed 2022-05-13 by instantlinux