instantlinux/docker-tools #80: Request: NUT 2.8.0

haxney, 2022-05-01:
NUT 2.8.0 [was released on April 26, 2022](https://github.com/networkupstools/nut/releases/tag/v2.8.0). I haven't been able to get NUT working with my SMART500RT1U, and I'm hoping the long-awaited 2.8.0 will help.

instantlinux, 2022-05-03:
I created a request in the alpine repo: track it there as [issue #13771](https://gitlab.alpinelinux.org/alpine/aports/-/issues/13771). Once it's published there, this image can be updated. Thanks for submitting the request!

instantlinux, 2022-05-11:
Version `2.8.0-r0` is now published.

closed 2022-05-13 by instantlinux

haxney, 2022-05-13:
Thanks! It turns out that the new version wasn't the problem with my setup; it was something with the container having access to USB. That means I've got more fighting to do with the configuration, but I'm glad to have the new version!

Thanks again for the super quick turnaround!