instantlinux/docker-tools #81: "This container may not work without SERIAL" then "Fatal error: 'serial' is not a valid variable name for this driver"

timdonovanuk, 2022-05-04:
Hi,

Wondering if you can help with nut-upsd.

I am not specifying SERIAL env var because it is not needed, but NUT spits out like it's still being sent:

```
Recreating nut-upsd_app_1 ... done
Attaching to nut-upsd_app_1
app_1  | ** This container may not work without setting for SERIAL **
app_1  | Network UPS Tools - UPS driver controller 3.15.0_alpha20210804-3402-gced1683082
app_1  | Network UPS Tools - APC Smart protocol driver 3.1 (3.15.0_alpha20210804-3402-gced1683082)
app_1  | APC command table version 3.1
app_1  |
app_1  | Fatal error: 'serial' is not a valid variable name for this driver.
app_1  |
app_1  | Look in the man page or call this driver with -h for a list of
app_1  | valid variable names and flags.
app_1  | Network UPS Tools upsd 3.15.0_alpha20210804-3402-gced1683082
app_1  | Ignoring invalid pid number 0
app_1  | listening on 0.0.0.0 port 3493
app_1  | Can't connect to UPS [ups] (apcsmart-ups): No such file or directory
app_1  | Network UPS Tools upsmon 3.15.0_alpha20210804-3402-gced1683082
app_1  |    0.000000    Ignoring invalid pid number 0
app_1  |    0.001201    Using power down flag file /etc/killpower
app_1  |    0.002002    Unable to use old-style MONITOR line without a username
app_1  |    0.002046    Convert it and add a username to upsd.users - see the documentation
app_1  |    0.002070    Fatal error: unusable configuration
```

Here is my docker-compose:

```
version: "3.1"

services:
  app:
    image: instantlinux/nut-upsd:latest
    restart: always
    environment:
      PORT: /dev/ttyUSB1
      DRIVER: apcsmart
    ports:
    - 3493:3493
    secrets:
    - nut-upsd-password
    privileged: true

secrets:
  nut-upsd-password:
    file: /var/adm/admin/secrets/nut-upsd-password
```

timdonovanuk, 2022-05-04:
I "fixed" it by mounting my own ups.conf but this is a bug, as it shouldn't send SERIAL if its not needed or defined. Cheers!

instantlinux, 2022-05-04:
_Very_ weird to hear this: APC is a leading UPS maker and this image has been working for a huge number of users for several years. Can you post your working ups.conf, so I can craft the correct update to `entrypoint.sh`? Thanks!

timdonovanuk, 2022-05-04:
Sure!

```
[apc-sc450]
    driver = apcsmart
    port = auto
    desc = "apc-sc450]"
```

instantlinux, 2022-05-05:
New image is posted, do a `docker pull` to get the update. I also updated the [Docker Hub README](https://hub.docker.com/r/instantlinux/nut-upsd) with info about how to get a list of available drivers and how to get the driver-specific help.

closed 2022-05-11 by instantlinux

timdonovanuk, 2022-05-11:
Thank you @instantlinux !