instantlinux/docker-tools #82: RPI 4 docker issue creating container Error

bsmith76s, 2022-05-11:
I am trying to create a container on raspberry Pi 4 running Bullseye OS 64 bit and keep getting this error 

2022-05-11T01:32:33.234710602Z Network UPS Tools - UPS driver controller 3.15.0_alpha20210804-3402-gced1683082
2022-05-11T01:32:33.236339183Z Network UPS Tools - Generic HID driver 0.41 (3.15.0_alpha20210804-3402-gced1683082)
2022-05-11T01:32:33.254306533Z Using subdriver: APC HID 0.96
2022-05-11T01:32:33.254487196Z USB communication driver 0.33
2022-05-11T01:32:33.369271592Z Network UPS Tools upsd 3.15.0_alpha20210804-3402-gced1683082
2022-05-11T01:32:33.369599104Z Ignoring invalid pid number 0
2022-05-11T01:32:33.371676288Z listening on 0.0.0.0 port 3493
2022-05-11T01:32:33.373039114Z Connected to UPS [apc]: usbhid-ups-apc
2022-05-11T01:32:33.388035835Z Network UPS Tools upsmon 3.15.0_alpha20210804-3402-gced1683082
2022-05-11T01:32:33.388223127Z    0.000000	Ignoring invalid pid number 0
2022-05-11T01:32:33.388704618Z    0.000489	Using power down flag file /etc/killpower
2022-05-11T01:32:33.388983724Z    0.000783	Unable to use old-style MONITOR line without a username
2022-05-11T01:32:33.389003854Z    0.000800	Convert it and add a username to upsd.users - see the documentation
2022-05-11T01:32:33.389106703Z    0.000921	Fatal error: unusable configuration

I have tried so many compose configurations here is the latest
version: "3.1"

services:
  app:
    image: instantlinux/nut-upsd:latest
    restart: always
    environment:
      SERIAL: 3B2109X63917
      NAME: apc
#      VENDERID: 0x051d
      SECRET: password
      DESCRIPTION: "APC Pro1500"
#      USER: Nutapc
#    devices:
#     - /dev/bus/usb/001/003
    ports:
    - 3493:3493
#    secrets:
#    - nut-upsd-password
    privileged: true

#secrets:
#  nut-upsd-password:
#    file: /var/adm/secrets/nut-upsd-password

Not sure what I am missing

instantlinux, 2022-05-11:
You're right, the doc wasn't clear; I've updated the [README](https://hub.docker.com/repository/docker/instantlinux/nut-upsd). There are two different env variables: `SECRET` and `API_PASSWORD`, you can specify the password with either one. To directly set the password, use the latter (and don't don't mount the volume `/run/secrets`); if you're defining a Docker secret to hold the value, mount the secrets volume and put the password into a secret named by the value of `SECRET`.

bsmith76s, 2022-05-11:
> You're right, the doc wasn't clear; I've updated the [README](https://hub.docker.com/repository/docker/instantlinux/nut-upsd). There are two different env variables: `SECRET` and `API_PASSWORD`, you [...]

Thank you I have container spun up now and logs say it is connected. I am fairly new at dockers I was trying to to use the console in portainer for the container to run the command upsc apc@localhost to see if it gave any data but it will not open to a console to run command. is this a permissions problem I need to address. or how do I know if I am getting any data from UPS?

closed 2022-05-18 by instantlinux