instantlinux/docker-tools #83: proftpd: restart loop on docker-compose due to adduser error

richardg867, 2022-06-28:
I run `instantlinux/proftpd` as part of a docker-compose stack, with `restart` set to `always`. Today, I found the server was no longer running, and docker-compose did not restart it as expected. Instead, the container's log had a flood of:

```adduser: user 'backup' in use```

...where `backup` is my configured `FTPUSER_NAME`. The entrypoint script should not attempt to create the FTP user if it somehow already exists.

instantlinux, 2022-08-30:
I don't see how this could happen. Here is the content of `/etc/passwd`:
```
# cat /etc/passwd
root:x:0:0:root:/root:/bin/ash
bin:x:1:1:bin:/bin:/sbin/nologin
daemon:x:2:2:daemon:/sbin:/sbin/nologin
adm:x:3:4:adm:/var/adm:/sbin/nologin
lp:x:4:7:lp:/var/spool/lpd:/sbin/nologin
sync:x:5:0:sync:/sbin:/bin/sync
shutdown:x:6:0:shutdown:/sbin:/sbin/shutdown
halt:x:7:0:halt:/sbin:/sbin/halt
mail:x:8:12:mail:/var/mail:/sbin/nologin
news:x:9:13:news:/usr/lib/news:/sbin/nologin
uucp:x:10:14:uucp:/var/spool/uucppublic:/sbin/nologin
operator:x:11:0:operator:/root:/sbin/nologin
man:x:13:15:man:/usr/man:/sbin/nologin
postmaster:x:14:12:postmaster:/var/mail:/sbin/nologin
cron:x:16:16:cron:/var/spool/cron:/sbin/nologin
ftp:x:21:21::/var/lib/ftp:/sbin/nologin
sshd:x:22:22:sshd:/dev/null:/sbin/nologin
at:x:25:25:at:/var/spool/cron/atjobs:/sbin/nologin
squid:x:31:31:Squid:/var/cache/squid:/sbin/nologin
xfs:x:33:33:X Font Server:/etc/X11/fs:/sbin/nologin
games:x:35:35:games:/usr/games:/sbin/nologin
cyrus:x:85:12::/usr/cyrus:/sbin/nologin
vpopmail:x:89:89::/var/vpopmail:/sbin/nologin
ntp:x:123:123:NTP:/var/empty:/sbin/nologin
smmsp:x:209:209:smmsp:/var/spool/mqueue:/sbin/nologin
guest:x:405:100:guest:/dev/null:/sbin/nologin
nobody:x:65534:65534:nobody:/:/sbin/nologin
proftpd:x:100:101:proftpd:/var/lib/ftp:/sbin/nologin
```
There is no `backup` user. So long as you specify `FTPUSER_NAME` and `FTPUSER_UID` with values other than any of the above, there shouldn't be any conflict. Please try again with version `1.3.7e-r0` and provide any other details (like a docker-compose file) needed to reproduce this issue.

instantlinux, 2022-08-30:
Figured out how to reproduce, posting a fix, thanks for the report!

closed 2022-11-25 by instantlinux