instantlinux/docker-tools #84: Samba: set_nt_acl_no_snum: fset_nt_acl returned NT_STATUS_ACCESS_DENIED on startup

nick-george, 2022-07-26:
Hi there, 

I'm trying out your image for the first time and am running into some issues getting it started. I have tried older images in case there was a bug in the latest, but that hasn't helped. It must be something I have done. Between test runs, I've deleted the volumes that were created, but that hasn't helped. I have also ensured that the DC is in DNS, and that the IP matches the host that this container is running on (but not its primary IP, it's a wireguard IP). 

Below is my compose file. 

```
services:
  samba:
    image: instantlinux/samba-dc:latest
    container_name: samba
    hostname: dc.testsite.co
    environment:
       INTERFACES: wg0 lo
       TZ: UTC
       REALM: dc.testsite.co
    network_mode: host
    restart: "no"
    secrets:
      - samba-admin-password

secrets:
  samba-admin-password:

```

Here are the logs coming out of the container at start time: 

```
user@host:~/work/samba$ docker-compose up
[+] Running 5/5
 ⠿ samba Pulled                                                                                                                                                                    20.8s
   ⠿ ba3557a56b15 Pull complete                                                                                                                                                     2.5s
   ⠿ 96c7c6ef7b74 Pull complete                                                                                                                                                     2.5s
   ⠿ b070a251e829 Pull complete                                                                                                                                                     2.5s
   ⠿ df3925289f41 Pull complete                                                                                                                                                    13.3s
[+] Running 1/1
 ⠿ Container samba  Created                                                                                                                                                         0.2s
Attaching to samba
samba  | Set timezone
samba  | INFO 2022-07-26 12:55:22,653 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2122: Looking up IPv4 addresses
samba  | WARNING 2022-07-26 12:55:23,014 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2132: No IPv4 address will be assigned
samba  | INFO 2022-07-26 12:55:23,014 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2139: Looking up IPv6 addresses
samba  | WARNING 2022-07-26 12:55:23,097 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2146: No IPv6 address will be assigned
samba  | INFO 2022-07-26 12:55:23,331 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2290: Setting up share.ldb
samba  | INFO 2022-07-26 12:55:23,369 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2294: Setting up secrets.ldb
samba  | INFO 2022-07-26 12:55:23,390 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2299: Setting up the registry
samba  | INFO 2022-07-26 12:55:23,471 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2302: Setting up the privileges database
samba  | INFO 2022-07-26 12:55:23,514 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2305: Setting up idmap db
samba  | INFO 2022-07-26 12:55:23,544 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #2312: Setting up SAM db
samba  | INFO 2022-07-26 12:55:23,552 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #897: Setting up sam.ldb partitions and settings
samba  | INFO 2022-07-26 12:55:23,553 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #909: Setting up sam.ldb rootDSE
samba  | INFO 2022-07-26 12:55:23,558 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1322: Pre-loading the Samba 4 and AD schema
samba  | Unable to determine the DomainSID, can not enforce uniqueness constraint on local domainSIDs
samba  | 
samba  | INFO 2022-07-26 12:55:23,593 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1400: Adding DomainDN: DC=dc,DC=testsite,DC=co
samba  | INFO 2022-07-26 12:55:23,615 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1432: Adding configuration container
samba  | INFO 2022-07-26 12:55:23,629 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1447: Setting up sam.ldb schema
samba  | INFO 2022-07-26 12:55:25,211 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1465: Setting up sam.ldb configuration data
samba  | INFO 2022-07-26 12:55:25,309 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1506: Setting up display specifiers
samba  | INFO 2022-07-26 12:55:26,544 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1514: Modifying display specifiers and extended rights
samba  | INFO 2022-07-26 12:55:26,565 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1521: Adding users container
samba  | INFO 2022-07-26 12:55:26,566 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1527: Modifying users container
samba  | INFO 2022-07-26 12:55:26,567 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1530: Adding computers container
samba  | INFO 2022-07-26 12:55:26,568 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1536: Modifying computers container
samba  | INFO 2022-07-26 12:55:26,568 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1540: Setting up sam.ldb data
samba  | INFO 2022-07-26 12:55:26,677 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1570: Setting up well known security principals
samba  | INFO 2022-07-26 12:55:26,702 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1584: Setting up sam.ldb users and groups
samba  | INFO 2022-07-26 12:55:26,796 pid:18 /usr/lib/python3.8/site-packages/samba/provision/__init__.py #1592: Setting up self join
samba  | Repacking database from v1 to v2 format (first record CN=MSMQ-Sign-Certificates,CN=Schema,CN=Configuration,DC=dc,DC=testsite,DC=co)
samba  | Repack: re-packed 10000 records so far
samba  | Repacking database from v1 to v2 format (first record CN=IntellimirrorSCP-Display,CN=404,CN=DisplaySpecifiers,CN=Configuration,DC=dc,DC=testsite,DC=co)
samba  | Repacking database from v1 to v2 format (first record CN=Machine,CN={6AC1786C-016F-11D2-945F-00C04FB984F9},CN=Policies,CN=System,DC=dc,DC=testsite,DC=co)
samba  | set_nt_acl_no_snum: fset_nt_acl returned NT_STATUS_ACCESS_DENIED.
samba  | ERROR(runtime): uncaught exception - (3221225506, '{Access Denied} A process has requested access to an object but has not been granted those access rights.')
samba  |   File "/usr/lib/python3.8/site-packages/samba/netcmd/__init__.py", line 186, in _run
samba  |     return self.run(*args, **kwargs)
samba  |   File "/usr/lib/python3.8/site-packages/samba/netcmd/domain.py", line 487, in run
samba  |     result = provision(self.logger,
samba  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 2341, in provision
samba  |     provision_fill(samdb, secrets_ldb, logger, names, paths,
samba  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1979, in provision_fill
samba  |     setsysvolacl(samdb, paths.netlogon, paths.sysvol, paths.root_uid,
samba  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1759, in setsysvolacl
samba  |     _setntacl(sysvol)
samba  |   File "/usr/lib/python3.8/site-packages/samba/provision/__init__.py", line 1753, in _setntacl
samba  |     return setntacl(
samba  |   File "/usr/lib/python3.8/site-packages/samba/ntacls.py", line 236, in setntacl
samba  |     smbd.set_nt_acl(
samba exited with code 255
```

Any ideas why this isn't working? 

Many thanks for your time, 
Nick

instantlinux, 2022-07-26:
That's pretty obscure: I can't help directly but the exception code 3221225506 shows up here - https://github.com/hierynomus/smbj/issues/328. Discussion there might help you track down what file it's trying to access.

instantlinux, 2022-08-11:
Did you find a solution?

nick-george, 2022-08-15:
Afraid not..  I had a look at the other issues, but they get into depth pretty quickly and I don't have the time to delve into trying out changes to the code/config in this project to try things out. 

I'm using this project with very little customisation, so I'm surprised that I've come across this issue (or that others haven't come across it). 

I'm happy to close the issue and call it a PEBKAC..

Kp0c, 2022-08-23:
try to add
```yaml
    cap_add:
      - SYS_ADMIN
```
It helped in my case

nick-george, 2022-08-24:
Thanks very much @Kp0c. That fixed it for me. Oh cr@p, I see it right there in the README to ensure this is added.. Doh!. 

Closing the ticket.

closed 2022-08-24 by nick-george