instantlinux/docker-tools #86: Keepalived does not remove VIP addr when it is stopped in docker

breezytm, 2022-09-19:
Dealing with this weird issue where keepalive is failing to remove the network from the master causing the load balancing to fail during a failover. I reached out to keepalived team and here's what they said is the root cause. Can you please advise? 

https://github.com/acassen/keepalived/issues/2201#issuecomment-1251444660

instantlinux, 2022-09-19:
Use `docker stop` before (or instead of) `docker down` (or run under Kubernetes) so the sigterm signal gets sent for graceful removal of the vip.

See this tutorial: https://www.sobyte.net/post/2022-05/docker-compose-graceful-stop/

breezytm, 2022-09-19:
It doesn't seem to make a difference unfortunately

breezytm, 2022-12-12:
> Use `docker stop` before (or instead of) `docker down` (or run under Kubernetes) so the sigterm signal gets sent for graceful removal of the vip.
> 
> See this tutorial: https://www.sobyte.net/post/2022-05/docker-compose-graceful-stop/

I was never able to get to the bottom of this issue but it is now clear to me it is because haproxy is being killed before keepalived hence the reason why the IP is not removed. If I execute a ``killall haproxy`` to the docker container, the IP address is removed. I have tried delaying the container stopping with different methods but never had any luck.

instantlinux, 2023-08-22:
The `2.8.2-alpine-2.2.8-r0` tag now supports SIGTERM when the container is invoked directly via docker-compose. My testing under kubernetes fails the same as before: the entrypoint script never sees the SIGTERM signal on my setup.

instantlinux, 2023-08-22:
Iterated a couple times, figured out that graceful stop under k8s is [not well documented](https://kubernetes.io/docs/concepts/workloads/pods/pod-lifecycle/#pod-termination) so I'm summarizing here for anyone else who searches for keywords `kubernetes`, `graceful stop`, `sigterm`, `sigusr1`, `not working` and the like:

* You need to define `STOPSIGNAL SIGTERM` in your Dockerfile
* Your application either needs to be running as pid 1 in the container, or you need to define a function in your entrypoint shell script that does this:
```
function graceful_stop() {
    echo "Received SIGTERM"
    kill -SIGTERM $(cat /run/keepalived/*.pid)
    sleep 1; exit 0
}
# ...
trap graceful_stop SIGTERM
# ...
wait
```
* After the function invokes SIGTERM to the subprocess(es), add a brief `sleep` so k8s doesn't stomp on the container before those subprocesses perform their own graceful shutdown
* If you're not using kubernetes, you need to define `stop_signal: SIGTERM` in your docker-compose file (or the equivalent command-line option to `docker`)

This turned out far more difficult to sort out than anyone should expect. Welcome to kubernetes.

The revised `2.8.2-alpine-2.2.8-r0` tag now behaves properly with or without kubernetes.

closed 2023-08-22 by instantlinux