instantlinux/docker-tools #94: Cannot read secret $ADMIN_PASSWORD_SECRET in /run/secrets

codedev900, 2022-11-15:
I'm trying to run the instantlinux/samba-dc docker image by using this command: docker run --detach --name devop --hostname test.ad.dev --privileged -p 389:389 instantlinux/samba-dc. The docker image will not start due to the following error: Cannot read secret $ADMIN_PASSWORD_SECRET in /run/secrets

instantlinux, 2022-11-16:
As noted in [README.md](https://github.com/instantlinux/docker-tools/blob/main/images/samba-dc/README.md), this image must be provided a passphrase: see the Secrets section. Provide the secret in a file mounted into the container as /run/secrets/samba-admin-password (if you're starting from `docker run` and have a local file named rootpw.txt, the syntax is `-v $(pwd)/rootpw.txt:/var/run/samba-admin-password`.)

codedev900, 2022-11-16:
Thank you for the quick reply. I added a file rootpw.txt and in the rootpw.txt I have samba-admin-password

I ran this command `docker run --detach --name devad --privileged -p 389:389 -v $(pwd)/rootpw.txt:/var/run/secrets instantlinux/samba-dc` but I'm still getting the same error "Cannot read secret $ADMIN_PASSWORD_SECRET in /run/secrets"

instantlinux, 2022-11-16:
OK do this:

* Put a random string like `f6npDxlqLikdb35XTtiW` into a file named rootpw.txt
* Specify the mounted file as `-v $(pwd)/rootpw.txt:/run/secrets/samba-admin-password` in your `docker run` command.

codedev900, 2022-11-16:
Thank you that worked. Now I'm using the LDAP Admin tool to login, but it's not working.

Base: DC=ad,DC=example,DC=com
Username: ad-example\administrator
Password: samba-admin-password

instantlinux, 2022-11-18:
The content of your rootpw.txt file is your _password_.
The value of environment variable `ADMIN_PASSWORD_SECRET` is the _docker secret_ name. By default that is `samba-admin-password`.

Why is it like this? So you can have two samba servers with different root passwords, from two or more different docker secrets.

codedev900, 2022-11-24:
Thank for your help!

closed 2022-11-24 by codedev900