instantlinux/docker-tools #95: samba-dc login issues (Kerberos) with Windows 11 22H2

nike12354, 2022-11-21:
Hi, 
can you please provide an updated samba version?
Starting with Win 11 22H2 client no login is possible anymore... See below.

--snip--
[image]
--snap--

RPCMoritz, 2022-11-25:
There was a new image published on Dockerhub three days ago, along with those commits, but it still reads

> ARG SAMBA_VERSION=4.15.7-r0

in the Dockerfile for that one.

What's needed to trigger the deloyment (and version-label) of the dockerfile on the current main branch featuring those commits?

RPCMoritz, 2022-11-25:
Ah, 15 minutes ago the new version appeared. Giving that a go :)

RPCMoritz, 2022-11-25:
Great success! [This](https://hub.docker.com/layers/instantlinux/samba-dc/4.16.7-r0/images/sha256-32748c8e8ca25396b679b9c908b10aac6ac7a5d2bba2062ada26fa2c085438e3?context=explore) fixes the login issues as expected, and booted up smoothly.

instantlinux, 2022-11-25:
Yes, what happened was alpine 3.17 was released, and my efforts to update from `edge` repo a couple days ago were a bit unstable. With 3.17, we now have a stable, updated version. Thanks for reporting this!

closed 2022-11-25 by instantlinux