instantlinux/docker-tools #107: Unable to run nut-upsd with an UPS connected by USB

masterlog80, 2023-03-19:
Hello,
After confirming the correct configuration on a bare metal machine, I am configuring instantlinux/nut-upsd as per the following settings on docker-compose (running on a X86_64 device):
```
  nut-upsd:
    container_name: nut-upsd
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    environment:
      - "API_USER=REDACTER"
      - "API_PASSWORD=REDACTER"
      - "DESCRIPTION=OMRON_BY50S"
      - "DRIVER=blazer_usb"
      - "GROUP=nut"
      - "NAME=BY50S"
      - "PORT=auto"
      - "SECRET=upspass"
      - "VENDORID=0590"
      - "PRODUCTID=0081"
      - "SUBDRIVER=ippon"
    devices:
      - /dev/bus/usb/002/002:/dev/bus/usb/002/002
    network_mode: host
    privileged: true
    restart: always
```
However, despite I have tried many configurations, it never gets connected to the UPS. 
On the container, I get the following messages:
```
# docker container logs nut-upsd -f
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

Driver failed to start (exit status=1)
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

Driver failed to start (exit status=1)
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).

Driver failed to start (exit status=1)
#
```

masterlog80, 2023-03-25:
Even using my running configuration by mounting as a volume, I got this error:
```
docker container logs nut-upsd -f
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
Failed to init libusb 1.0: No such file or directory
Driver failed to start (exit status=1)
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
Failed to init libusb 1.0: No such file or directory
Driver failed to start (exit status=1)
```
It seems missing a library for the USB.

instantlinux, 2023-03-25:
Thanks for the clarification, the driver for this hardware (blazer / megatec) apparently requires the `libusb` alpine package. It's added now, please invoke
```
docker pull instantlinux/nut-upsd:2.8.0-r4
```
and try your use-case again.

masterlog80, 2023-03-26:
Hello,
Thank you for your reply and effort. If my configuration above is correct (and it should, as I have created a new VM which works) the container does still have (ideally the same) issues whatever I set for `PORT`:
```
Driver failed to start (exit status=1)
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Megatec/Q1 protocol USB driver 0.14 (2.8.0)
WARNING: warn_if_bad_usb_port_filename(): port argument specified to
  the driver is "/dev/bus/usb/002/007" but USB drivers do not use it and rely on
  libusb walking all devices and matching their identification metadata.
  NUT documentation recommends port="auto" for USB devices to avoid confusion.
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 blazer_usb).
Driver failed to start (exit status=1)
```

To better clarify, I had this working on an Alpine standalone VM by following (and a little changing) those instructions:
https://www.fedux.org/articles/monitor-a-usb-connected-uninterruptible-power-supply-with-alpine-linux.html
Hope it can help!

instantlinux, 2023-03-26:
I think the `libusb` package may have nothing to do with this, then. That article talks about setting up udev rules for permissions (on the host server). The container runs nut as user `nut` with uid 100 and gid 101 (at present - this could change but probably hasn't in the years since I have maintained it). To troubleshoot permissions, invoke `docker exec -it <id> sh` and do `ls -lR /dev/bus/usb`. Also do `lsusb` within the container (it will only show the device IDs without their descriptions). On the host, invoke `lsusb` to see which device ID is your UPS. The device node that matches (under /dev/bus/usb) has to be readable by uid 100 or gid 101. If it isn't, set up a udev rule to make it group-readable for that gid.

I'm assuming you've already checked all that but it's worth sharing what you've learned here.

masterlog80, 2023-03-26:
Hello @instantlinux thank you for your reply.
Yeah, I have already checked and the USB port of the UPS is available inside the container as well, and matching the one on the system. Also, if I run `cat` on it, I can see the same character string I see on the **host** so IMHO this may not be the issue.
Actually I am running NUT in a small VM with Alpine modified per the instructions above and it works well, but I am thinking to try within a LXC container as well. Another workaround would be using an OS different from Alpine, if I can find something lightweight.

Thanks

LuisPeregrina, 2025-01-26:
Regressed, this issue is present in latest (a15f20ef0e2ae64e7f7cd1622425f5f1f673fd91):
```
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic Q* USB/Serial driver 0.36 (2.8.2)
Failed to init libusb 1.0: No such file or directory
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
USB communication driver (libusb 1.0) 0.47
Driver failed to start (exit status=1)
```

awushensky, 2025-03-17:
I'm having the same issue described above.

```
$ sudo docker logs -f nut-servers

Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
USB communication driver (libusb 1.0) 0.47
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
USB communication driver (libusb 1.0) 0.47
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
interrupt pipe disabled (add 'pollonly' flag to 'ups.conf' to get rid of this message)
libusb1: Could not open any HID devices: insufficient permissions on everything
No matching HID UPS found
upsnotify: failed to notify about state 4: no notification tech defined, will not spam more about it
USB communication driver (libusb 1.0) 0.47
Driver failed to start (exit status=1)
```

I checked the readability of the APC UPS within the docker container
```
$ sudo docker exec -it nut-servers lsusb

...
Bus 005 Device 002: ID 051d:0003 American Power Conversion Smart-UPS 2200 FW:UPS 09.3 / ID=18
...
```

and the permissions
```
$ sudo docker exec -it nut-servers ls -lR /dev/bus/usb

...
/dev/bus/usb/005:
total 0
crw-rw-r--    1 root     root      189, 512 Mar 17 05:28 001
crw-rw-r--    1 root     root      189, 513 Mar 17 05:28 002
...
```

I wasn't able to launch a shell on the container since it appears to be crashing and re-creating on a fast loop. I'm not sure how long this has been going on, I realized my monitoring has been broken for a while.

My docker compose config is pretty minimal:
```
  nut_servers:
    image: instantlinux/nut-upsd:latest
    container_name: nut-servers
    restart: unless-stopped
    labels:
      - "diun.enable=true" # enable update notifications
    environment:
      SERIAL: WS1211180096
    ports:
      - 3493:3493
    secrets:
      - nut-upsd-password
    privileged: true
    networks:
      nut: # connection to nut-web and nut client internally
```

instantlinux, 2025-03-28:
Cross-linking to the upstream: https://github.com/networkupstools/nut/issues/2447 is where discussion of this issue (as summarized above by @LuisPeregrina and @awushensky) is taking place.

awushensky, 2025-04-28:
following-up here, I'm embarrassed to admit that the issue I was experiencing was due to a faulty USB cable. It turns out the connection between my server and the UPS had gone bad.

instantlinux, 2025-05-02:
OK thanks. I'll close this out: so far as I can tell, any remaining issues are with the `nut` package published upstream. Please open a new issue here for any new symptoms.