instantlinux/docker-tools #109: nut-upsd: Out of memory

Solarunit, 2023-04-05:
I have a problem with rinning nut-upsd, in the logs is see:

`Out of memory: Out of memory`

I guess it's related to this:
https://github.com/networkupstools/nut/issues/1672

As workaround there was mention this: 

`/usr/sbin/upsd -F &`

I was trying to add this in my docker-compose.yml by adding `entrypoint:` and `command:` but no luck so far..

instantlinux, 2023-09-05:
Please provide more details about the hardware, software and configuration: how can we reproduce this elsewhere?

myakove, 2023-12-24:
Same here.
UPS: Eaton UPS 5E 1500i (usbhid-ups)

```
nut-upsd:
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    hostname: nut-upsd
    restart: unless-stopped
    privileged: true
    ports:
      - 3493:3493/tcp
    environment:
      - API_USER=user
      - API_PASSWORD=pass
      - NAME=UPS
      - DESCRIPTION=UPS Eaton 5E 1500i
      - VENDORID=0463
      - POLLINTERVAL=1
```
Got error: `Out of memory: Out of memory`

For now, got it running by modify `entrypoint.sh` and build locally.
Changed `/usr/sbin/upsd -u $USER` >> `/usr/sbin/upsd -u $USER -F &`

I have 3 PCs with `Eaton` UPSs, only on one of them the change was needed.

All of them use the same docker version:
```
docker version
Client: Docker Engine - Community
 Version:           24.0.7
 API version:       1.43
 Go version:        go1.20.10
 Git commit:        afdd53b
 Built:             Thu Oct 26 09:09:56 2023
 OS/Arch:           linux/amd64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          24.0.7
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.10
  Git commit:       311b9ff
  Built:            Thu Oct 26 09:08:22 2023
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.6.26
  GitCommit:        3dd1e886e55dd695541fdcd67420c2888645a495
 runc:
  Version:          1.1.10
  GitCommit:        v1.1.10-0-g18a0cb0
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

Let me know if more info is needed.

Solarunit, 2023-12-24:
>Please provide more details about the hardware, software and configuration: how can we reproduce this elsewhere?

I am sorry, I can longer reproduce it because i don't use nut docker now.

instantlinux, 2024-12-09:
Closing stale ticket.

closed 2024-12-09 by instantlinux