instantlinux/docker-tools #111: Add SFTP support to proftpd

c0da, 2023-04-20:
With this commit, two environment variables are added to enable and use SFTP instead of FTP.

The variable SFTP_ENABLE on/off (default off), installs the proftpd-mod_sftp and openssh-keygen packages, generates the ssh keys, and configures proftpd to use only SFTP_PORT (default 2222) instead of 21.

It also includes a change to disable anonymous login, which deletes the "<Anonymous>" section of the configuration instead of changing the RequireValidShell variable to on/off, since proftpd will not start if it has the <Anonymous> section set.

c0da, 2023-04-21:
Just out of curiosity, it seems that the EXPORT values in the Dockerfile are only taken into account at the build stage. When running the image, it doesn't matter what values are used in the PASV_MIN_PORT and PASV_MAX_PORT environment variables. With a "docker ps", the ports shown as exposed in the "PORTS" column are always the same (the default ones).

instantlinux, 2023-04-24:
Tag `1.3.7f-r1` now has this capability. Please verify, and thanks for the contribution!
```
docker pull instantlinux/proftpd:1.3.7f-r1
```

c0da, 2023-04-25:
Thank you @instantlinux !

I think I have found a small detail. When SFTP_ENABLE is set to on, the value of ANONYMOUS_DISABLE is still off (its default), so the \<Anonymous\> section is not removed from the configuration. Anyway, proftpd works without any issues, but gives the warning saying that the \<Anonymous\> section is going to be ignored by mod_sftp. Sorry for the mistake.

I was thinking that an alternative would be to set ANONYMOUS_DISABLE to "on" (in the entrypoint.sh file) inside the check that verifies that SFTP_ENABLE is also "on".

Do you think it would be necessary to create a new PR to correct this?

c0da, 2023-05-08:
> I was thinking that an alternative would be to set ANONYMOUS_DISABLE to "on" (in the entrypoint.sh file) inside the check that verifies that SFTP_ENABLE is also "on".

I will open a new PR for this.

closed 2023-05-08 by c0da