instantlinux/docker-tools #114: Workaround for July 2023 Windows Update

mstarostik, 2023-07-17:
The July 2023 Windows Update breaks cooperation with Samba servers.
Patch is available in https://bugzilla.samba.org/show_bug.cgi?id=15418
Any chance to get this into instantlinux/samba-dc soonish?

instantlinux, 2023-07-17:
OK, as soon as it gets into the 4.18.x patches in alpine. The upstream got merged for [debian](https://bugzilla.samba.org/show_bug.cgi?id=15418#c61) only a few hours ago.

instantlinux, 2023-07-25:
Delayed deployment by this at gitlab:
```
No more CI minutes available
You have used 448 out of 400 of your shared Runners pipeline minutes. For more information, go to the [Runners page.](https://gitlab.com/help/ci/runners/configure_runners.html#set-maximum-job-timeout-for-a-runner)
```

instantlinux, 2023-07-26:
Tag [4.18.5-r0](https://hub.docker.com/layers/instantlinux/samba-dc/4.18.5-r0/images/sha256-81e09db6ac39fec10e0154fb274c3e7e896ac7ff2f258c1257f9abc36cb2ba1c?context=repo) is now published, based on the alpine package published 20-Jul. Please confirm that this fixes the issue, @mstarostik - thanks!

mstarostik, 2023-08-09:
Yes, that seems to do it. Thank you!

closed 2023-08-09 by mstarostik