instantlinux/docker-tools #128: nut-upsd - telegram support

S474N, 2023-09-01:
I have one idea for improvement - Telegram support, for example for sending state of UPS (without power, etc.).

instantlinux, 2023-09-02:
The NUT Tools package from which this container is built doesn't have built-in notifications: it provides an API for monitoring, via TCP port 3493.

A typical alert system is Nagios, which has a few rivals such as OpGenie, Zabbix and/or cloud services like DataDog. If you run [nagios](https://hub.docker.com/r/instantlinux/nagios), you can use the standard [check_ups plugin](https://nagios-plugins.org/doc/man/check_ups.html) and set up notifications to any medium you choose, such as the [telegram integration for nagios](https://exchange.nagios.org/directory/Addons/Notifications/Telegram-Notifications/details).

S474N, 2023-09-02:
Why so complicated?

In persistent storage we have file `upsmon.conf`. Here we can define `NOTIFYCMD`  (and another commands). 

Simple SH script:
```
#!/bin/sh

hostname=`hostname`

curl -s \
  --data parse_mode=HTML \
  --data chat_id=XXXXXX \
  --data text="<b>nut-upsd </b>%0A      <i>sending from <b>#RPi4B</b></i>%0A%0A${1}" \
  "https://api.telegram.org/botXXXXX:YYY/sendMessage" \
  > /dev/null 2>&1

exit 0
```

But have problem with permision of this file :(

```
28. 08. 2023 5:44:55 sh: /etc/nut/local/send_telegram_message.sh: Permission denied
28. 08. 2023 15:44:25 253948.886598	UPS apc750@localhost - jede z baterii!
28. 08. 2023 15:44:25 sh: /etc/nut/local/send_telegram_message.sh: Permission denied
28. 08. 2023 15:44:50 253973.891283	UPS apc750@localhost - obnoven privod elektriny.
28. 08. 2023 15:44:50 sh: /etc/nut/local/send_telegram_message.sh: Permission denied
31. 08. 2023 5:35:42 476626.527492	UPS eaton650@localhost - jede z baterii!
```

instantlinux, 2023-09-03:
Please open a pull request rather than an Issue if you want this suggestion implemented.

S474N, 2023-09-04:
Email is default in UPS settings. I don't know, if Twitter/X has API for this purposes. 

As I wrote above, I have still problem with `permissions denied`, although script has 777 privileges and owned by root user (tested also pi).

instantlinux, 2023-09-04:
In your testing, check the ownership and permissions of the shell script as well as the directory holding it (you created `/etc/nut/local`). In the published container, I see the config files under `/etc/nut` set to 640 permissions, owner `root`, group `nut`. Since APIs require passwords or secure tokens, this interface script needs to reference environment variables that are passed to the container as secrets.

S474N, 2023-09-04:
OK, copied script to `/etc/nut` directly in container and set owner:
[image]

And I will test it :)

instantlinux, 2025-09-22:
Revisiting this 2 years later, in context of similar user requests: support for `NOTIFYCMD` is (finally!) added as of tag `2.8.3-r2`.

closed 2025-09-22 by instantlinux