instantlinux/docker-tools #129: Can't claim USB device / Entity not found

Piscator74, 2023-09-01:
I am trying to set up an UPS with the following yml 

```
version: '3.1'

services:
  nut-uspd:
    container_name: nut-upsd
    image: instantlinux/nut-upsd:2.8.0-r4
    restart: always
    ports:
      - 3494:3493
    environment:
      API_USER: upsmon
      API_PASSWORD: nosecret
      SERIAL: 5B2202T37802
      PORT: auto
    devices:
      - /dev/ttyUSB2:/dev/ttyUSB2
    privileged: true
```

lsusb anb lsusb -v give the following information about the UPS
```
Bus 002 Device 009: ID 051d:0002 American Power Conversion Uninterruptible Power Supply
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x051d American Power Conversion
  idProduct          0x0002 Uninterruptible Power Supply
  bcdDevice            0.90
  iManufacturer           1 American Power Conversion
  iProduct                2 Back-UPS ES 850G2 FW:938.a2 .I USB F
  iSerial                 3 5B2202T37802  
  bNumConfigurations      1
<snip>
```

But I keep getting the following error in the docker logs. Any clue what is going wrong here?
```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
Can't claim USB device [051d:0002]@0/0: Entity not found
USB communication driver (libusb 1.0) 0.43
Driver failed to start (exit status=1)
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
Can't claim USB device [051d:0002]@0/0: Entity not found
Driver failed to start (exit status=1)
```

instantlinux, 2023-09-01:
Looks like permissions of a device on the host. You have the port set to `auto`, so this section shouldn't be necessary:
```
    devices:
      - /dev/ttyUSB2:/dev/ttyUSB2
```
Such a device name shouldn't exist on the host. Maybe there is another driver installed on the host which is grabbing the usb device `051d:0002` and attaching it to a dev inode like this? Look for that driver and disable it.

Piscator74, 2023-09-02:
Thanks for the suggestion! The host was indeed locking the driver. It works now and I can read the status of the UPS.

closed 2023-09-02 by Piscator74