instantlinux/docker-tools #134: [SAMBA_DC] How can I upload data in ldif (or any other) format?

gredwhite, 2023-10-03:
I use following docker-compose file:

```
version: '3.4'

services:
  samba:
    image: instantlinux/samba-dc:latest
    container_name: samba-dc
    cap_add:
      - CAP_SYS_ADMIN
    hostname: my.company
    environment:
      DOMAIN_ACTION: provision
      REALM: inno.tech
    volumes:
      - etc:/etc/samba
      - lib:/var/lib/samba
    ports:
      - "389:389"
    secrets:
      - samba-admin-password

volumes:
  etc:
  lib:

secrets:
  samba-admin-password:
    file: secrets.yaml
```
I have ldif file and I want to upload it to server ? Is there way to do it ? I tried to do it via **samba-tool** but no luck. please help

instantlinux, 2023-10-25:
To transfer data, you can join a samba-dc to an existing, running Active Directory controller. So far as I know, from samba-tool there isn't a way to import an ldif backup file exported by a no-longer-existing, shut-down Active Directory controller. If I were trying to do this, I'd look at other tools like Apache Directory Studio or perhaps the Microsoft AD manager: they can import / export ldif files.

instantlinux, 2025-09-25:
Closing due to inactivity.

closed 2025-09-25 by instantlinux