instantlinux/docker-tools #138: [Question] Is it possible to change the POLLFREQ on the docker image?

sblantipodi, 2023-11-04:
As title.
I'm using the docker image:

https://hub.docker.com/r/instantlinux/nut-upsd

when I run this command:
`upsc Cyberpower_PR1500ELCD@localhost`

I get this output:
```
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 5730
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: PR1500ELCD
device.serial: PRTGN2000005
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/bus/usb/001/003
driver.parameter.synchronous: auto
driver.version: 2.8.0
driver.version.data: CyberPower HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.voltage: 234.0
input.voltage.nominal: 230
output.voltage: 265.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 17
ups.mfr: CPS
ups.model: PR1500ELCD
ups.productid: 0601
ups.realpower.nominal: 1350
ups.serial: PRTGN2000005
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```
Can I change the pollfreq that is set at 30?

Even if my pollinterval is set at 2, it seems that I get only 1 new reading every 30 seconds...

Thanks!

instantlinux, 2023-11-06:
As noted in the [README](https://github.com/instantlinux/docker-tools/blob/main/images/nut-upsd/README.md), you can override any of the configuration files by volume-mounting them under `/etc/nut/local`. So if you want to change `POLLFREQ`, make a copy of the `/etc/nut/upsmon.conf` file, modify this setting, and assign a docker volume mount of the updated file as `/etc/nut/local/upsmon.conf`.

This will require adding a volume directive to the helm chart or docker-compose.yml you use for launching the app.

sblantipodi, 2023-11-06:
hi @instantlinux I have done what you suggested.

if I attach to the container I can see the upsmon.conf mounted in /etc/nut/local and it contains the following:

```
/ # cat /etc/nut/local/upsmon.conf 
MONITOR Cyberpower_PR1500ELCD@localhost 1 admin mypwd master
RUN_AS_USER nut
POLLFREQ 5
```

but the pollfreq doesn't change, my UPS readings are updated every 30 seconds... 
any idea on a possible solution to this problem? thanks!!!

sblantipodi, 2023-11-07:
ops, I solved it by editing the ups.conf file and not the upsmon.conf
thank you!!!!

closed 2023-11-07 by sblantipodi