instantlinux/docker-tools #171: Health check kill arguments not valid

keyboardr, 2024-12-03:
I got the data stale issue and the health check didn't flag it. I opened a console in the container while it was in that state and ran the command and got `kill: Illegal option -S`.

I think changing the kill portion to `kill -s SIGTERM -1` should fix it. I've updated my compose to override it to that.

keyboardr, 2024-12-03:
It didn't like that either, so I just changed the whole thing to `upsc ups@localhost:3493 2>&1|grep -q battery.charge`, which doesn't drop the process. Instead I'll rely on a `deunhealth` container to manage the restart.

instantlinux, 2024-12-06:
A syntax that works is `killall -TERM upsmon`. Alpine 3.21 hasn't come out yet, so publishing a new image is a bit tricky with reliance on `edge` repo (some of the packages there shifted since last image build), which is currently required to pass security-vulnerability checks. Once I sort that out, there will be an image `2.8.2-r2` or later.

instantlinux, 2024-12-09:
I've corrected the command and posted this fix with the `v2.8.2-r2` update. Let me know if there are further issues.

closed 2024-12-15 by instantlinux