instantlinux/docker-tools #175: nut-upsd - Is there a way to shutdown the host ?

Elu43, 2024-12-29:
Hi
Is there a way to act as a client as well, and shutdown the host when UPS battery is low ?

Best regards

instantlinux, 2025-01-29:
I googled around in search of an answer for you, and came up empty. There are references to setting up an passwordless ssh key-pair, to set this in upsmon.conf:
`SHUTDOWNCMD "ssh k8s-worker1 sudo shutdown -h now"`

But there isn't a single page I can find the details. This image doesn't have openssh-client installed, so if you take that route, you'll need to override the entrypoint to invoke `apk add --update openssh-server && /usr/local/bin/entrypoint.sh` at startup, and volume-mount the required ssh key into /var/.ib/nut/.ssh.

instantlinux, 2025-03-28:
Closing due to inactivity: comments welcome here for the benefit of anyone else who requires this capability.

closed 2025-03-28 by instantlinux