instantlinux/docker-tools #210: Samba start error

tonisimakov99, 2025-07-14:
samba in docker not starts with error:
```INFO 2025-07-14 03:40:34,282 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1384: Adding DomainDN: DC=samba,DC=com

INFO 2025-07-14 03:40:34,323 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1416: Adding configuration container

INFO 2025-07-14 03:40:34,355 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1431: Setting up sam.ldb schema

INFO 2025-07-14 03:40:35,648 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1451: Setting up sam.ldb configuration data

INFO 2025-07-14 03:40:35,743 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1493: Setting up display specifiers

INFO 2025-07-14 03:40:36,773 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1501: Modifying display specifiers and extended rights

INFO 2025-07-14 03:40:36,791 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1508: Adding users container

INFO 2025-07-14 03:40:36,793 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1514: Modifying users container

INFO 2025-07-14 03:40:36,794 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1517: Adding computers container

INFO 2025-07-14 03:40:36,795 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1523: Modifying computers container

INFO 2025-07-14 03:40:36,796 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1527: Setting up sam.ldb data

INFO 2025-07-14 03:40:36,874 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1558: Setting up well known security principals

INFO 2025-07-14 03:40:36,893 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1572: Setting up sam.ldb users and groups

INFO 2025-07-14 03:40:36,984 pid:17 /usr/lib/python3.11/site-packages/samba/provision/__init__.py #1580: Setting up self join

Repacking database from v1 to v2 format (first record CN=ms-WMI-RealRangeParam,CN=Schema,CN=Configuration,DC=samba,DC=com)

Repack: re-packed 10000 records so far

Repacking database from v1 to v2 format (first record CN=server-Display,CN=415,CN=DisplaySpecifiers,CN=Configuration,DC=samba,DC=com)

Repacking database from v1 to v2 format (first record CN=ComPartitionSets,CN=System,DC=samba,DC=com)

ERROR(<class 'samba.provision.ProvisioningError'>): Provision failed - ProvisioningError: Your filesystem or build does not support posix ACLs, which s3fs requires.  Try the mounting the filesystem with the 'acl' option.
```

What could be the problem?

instantlinux, 2025-09-20:
Please provide more detail about the volumes attached to your container: your docker-compose or helm chart. The volume in question is (I assume) /var/lib/samba: is it `ext4` or some other filesystem type?

instantlinux, 2025-09-24:
Samba domain controller requires a POSIX filesystem, not an S3 block-store or other type of volume. Mount /var/lib/samba from (Linux) block storage volume or an NFS fileserver.