instantlinux/docker-tools #213: "Unable to enter background: Out of memory" on Debian Trixie Host

Certain1989, 2025-08-10:
First of all: thank you for the Software. I seem to have stumbled on a Problem. When I was using Debian Bookworm as a Docker host, it worked very well. After the Host was updated to Trixie, the container will not start and the logs show an out of memory error when trying to shuft to background:

```bash
Network UPS Tools - UPS driver controller 2.8.2
Network UPS Tools - Generic HID driver 0.53 (2.8.2)
Using subdriver: MGE HID 1.46
USB communication driver (libusb 1.0) 0.47
Network UPS Tools upsd 2.8.2
Ignoring invalid pid number 0
listening on 0.0.0.0 port 3493
Connected to UPS [EatonUPS]: usbhid-ups-EatonUPS
Found 1 UPS defined in ups.conf
Unable to enter background: Out of memory
```
My compose looks like this:

```bash
nut-upsd:
    image: instantlinux/nut-upsd
    container_name: nut-upsd
    environment:
      - TZ=**********
      - API_USER=NUTAPI
      - API_PASSWORD=***************************
      - DESCRIPTION=EatonEllipseEco650
      - DRIVER=usbhid-ups
      - NAME=EatonUPS
      - SERIAL=000000000
    devices:
      - /dev/bus/usb:/dev/bus/usb
    networks:
      - internal
    ports:
      - 3493:3493
    restart: unless-stopped
```
Is there something I have to change to get the container to a good state on Trixie?

Thanks in advance.

adrienyhuel, 2025-08-11:
@Certain1989 

I got it working with comment : https://github.com/networkupstools/nut/issues/1672#issuecomment-2303260972

I added the ulimit parameter to my docker compose, and it works

Certain1989, 2025-08-11:
@adrienyhuel 

Thank you, that did the trick. Mine is now working as well. Might be something that will have to go into the official container, built in or as instructions for the docker compose configuration.

Have a nice remaining day.

HanaO00, 2025-08-28:
Thank you it worked for me also

instantlinux, 2025-08-30:
Please try the new tag `2.8.3-r2`. You can remove the docker-compose parameter: default ulimit for files is set to 2048 as an environment variable. Rather than docker-compose, this variable is referenced in entrypoint so it will work under kubernetes as well as docker-compose.

closed 2025-08-30 by instantlinux